quickreflex/rtl8188eus #1: This driver does not support the monitor mode

ivakor644, 2018-05-12:
Could you implement this case so that monitor mode support works, at least for Kali Linux. The chip I have is RTL8188EUS, and wifi is a usb wifiagent card. Here it is: http://4pda.ru/2016/11/21/329596 or here you can see: https://remo-zavod.ru/katalog/communication-antennas-and-equipment/antennas-wifi/antenna-wifi -agent. I will be very grateful! Generally did this instruction: https://forums.kali.org/showthread.php?37911-Getting-RTL8188-to-work-with-Kali-in-monitor-mode, but the kernel headers are not suitable. Here is the conclusion of airmon-ng: Realtek Semiconductor Corp. RTL8188EUS 802.11n Wireless Network

ivakor644, 2018-05-12:
And if you put this driver: https://github.com/lwfinger/rtl8188eu, the output of airmon-ng will be as follows:
```
root@debian:~# airmon-ng

PHY	Interface	Driver		Chipset

phy0	wlan0		iwlwifi		Intel Corporation Wireless 7265 (rev 61)
null	wlan1		r8188eu		Realtek Semiconductor Corp. RTL8188EUS 802.11n Wireless Network Adapter
```
```
root@debian:~# airmon-ng start wlan1

Found 3 processes that could cause trouble.
If airodump-ng, aireplay-ng or airtun-ng stops working after
a short period of time, you may want to run 'airmon-ng check kill'

  PID Name
  623 NetworkManager
  740 wpa_supplicant
  787 dhclient

PHY	Interface	Driver		Chipset

phy0	wlan0		iwlwifi		Intel Corporation Wireless 7265 (rev 61)
null	wlan1		r8188eu		Realtek Semiconductor Corp. RTL8188EUS 802.11n Wireless Network Adapter
```

ivakor644, 2018-05-12:
```
root@debian:~# iw dev
phy#1
	Interface wlan1
		ifindex 4
		wdev 0x100000001
		addr 00:13:ef:81:01:8f
		type managed
	        txpower 12.00 dBm
```

```
root@debian:~# iw list
Wiphy phy1
	max # scan SSIDs: 9
	max scan IEs length: 2304 bytes
	max # sched scan SSIDs: 0
	max # match sets: 0
	max # scan plans: 1
	max scan plan interval: -1
	max scan plan iterations: 0
	Retry short limit: 7
	Retry long limit: 4
	Coverage class: 0 (up to 0m)
	Supported Ciphers:
		* WEP40 (00-0f-ac:1)
		* WEP104 (00-0f-ac:5)
		* TKIP (00-0f-ac:2)
		* CCMP-128 (00-0f-ac:4)
	Available Antennas: TX 0 RX 0
	Supported interface modes:
		 * IBSS
		 * managed
		 * AP
		 * monitor
		 * P2P-client
		 * P2P-GO
	Band 1:
		Capabilities: 0x1862
			HT20/HT40
			Static SM Power Save
			RX HT20 SGI
			RX HT40 SGI
			No RX STBC
			Max AMSDU length: 7935 bytes
			DSSS/CCK HT40
		Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
		Minimum RX AMPDU time spacing: 16 usec (0x07)
		HT Max RX data rate: 150 Mbps
		HT TX/RX MCS rate indexes supported: 0-7, 32
		Bitrates (non-HT):
			* 1.0 Mbps
			* 2.0 Mbps
			* 5.5 Mbps
			* 11.0 Mbps
			* 6.0 Mbps
			* 9.0 Mbps
			* 12.0 Mbps
			* 18.0 Mbps
			* 24.0 Mbps
			* 36.0 Mbps
			* 48.0 Mbps
			* 54.0 Mbps
		Frequencies:
			* 2412 MHz [1] (20.0 dBm)
			* 2417 MHz [2] (20.0 dBm)
			* 2422 MHz [3] (20.0 dBm)
			* 2427 MHz [4] (20.0 dBm)
			* 2432 MHz [5] (20.0 dBm)
			* 2437 MHz [6] (20.0 dBm)
			* 2442 MHz [7] (20.0 dBm)
			* 2447 MHz [8] (20.0 dBm)
			* 2452 MHz [9] (20.0 dBm)
			* 2457 MHz [10] (20.0 dBm)
			* 2462 MHz [11] (20.0 dBm)
			* 2467 MHz [12] (20.0 dBm) (no IR)
			* 2472 MHz [13] (20.0 dBm)
			* 2484 MHz [14] (20.0 dBm) (no IR)
	Supported commands:
		 * new_interface
		 * set_interface
		 * new_key
		 * start_ap
		 * new_station
		 * set_bss
		 * join_ibss
		 * set_pmksa
		 * del_pmksa
		 * flush_pmksa
		 * remain_on_channel
		 * frame
		 * set_channel
		 * connect
		 * disconnect
	Supported TX frame types:
		 * IBSS: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * managed: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * AP: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * AP/VLAN: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * P2P-client: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * P2P-GO: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
	Supported RX frame types:
		 * IBSS: 0xd0
		 * managed: 0x40 0xd0
		 * AP: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
		 * AP/VLAN: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
		 * P2P-client: 0x40 0xd0
		 * P2P-GO: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
	WoWLAN support:
		 * wake up on anything (device continues operating normally)
	software interface modes (can always be added):
		 * monitor
	interface combinations are not supported
	Device supports scan flush.

```

ivakor644, 2018-05-12:
И вот теперь при переводе в режим мониторинга показывает такую ошибку
`Error setting channel: command failed: Operation not supported (-95)`

quickreflex, 2018-05-28:
The driver in this mode does not work at this stage. I need to change a large amount of code to work. I can try, but it will take a long time.

ivakor644, 2018-05-29:
yes, if you could remake the driver under monitor mode, I'm sure that I would be grateful to you not only for me! After all, this is the most common problem for this manufacturer. Almost all Realtek chips do not initially support the monitor mode out of the box. But still the decision is yours!

kimocoder, 2018-07-17:
The drivers doesn't support virtual interface, that's why airmon-ng won't work and it has to be set in monitor mode manually.

seksitha, 2018-08-26:
Hi, I'm new to this. my Kali kernel is 4.17. I get only auto mode by doing:
ifconfig wlan1 down
iwconfig wlan1 mode monitor
ifconfig wlan1 up
It can not do deauth. :(. 
any help, please?

GHXST01, 2018-12-03:
I think this issue can be closed? Maybe a separate issue tracker for virtual interfaces?

sanyam8929, 2020-07-27:
same problem ..
pls find some solution

quickreflex, 2022-11-10:
Prueba esto:     sudo iw dev wlan0 set type monitor  

 >-------- Оригинално писмо --------

 >От: MelmacCode ***@***.***

 >Относно: Re: [quickreflex/rtl8188eus] This driver does not support the monitor
 mode (#1)

 >До: quickreflex/rtl8188eus  

 >Изпратено на: 10.11.2022 02:40

 
 
  
  
  
a mi me presenta este error al querer activar el modo monitor de forma manual
 ┌──(root㉿kali)-[~]
 └─# ifconfig wlan0 down  
  
┌──(root㉿kali)-[~]
 └─# iwconfig wlan0 mode monitor
 Error for wireless request "Set Mode" (8B06) :
 SET failed on device wlan0 ; Invalid argument.  
  
Alguien tendra alguna solución..?