quickreflex/rtl8188eus #8: Compile failing

EvilOlaf, 2018-08-06:
```
root@orangepioneplus:~/rtl8188eus# make all
"******************************************"
"NO SKRC,we will use default KSRC"
"******************************************"
make ARCH=arm64 CROSS_COMPILE= -C /lib/modules/4.17.11-sunxi64/build M=/root/rtl8188eus  modules
make[1]: Entering directory '/usr/src/linux-headers-4.17.11-sunxi64'
"******************************************"
"NO SKRC,we will use default KSRC"
"******************************************"
  CC [M]  /root/rtl8188eus/core/rtw_cmd.o
  CC [M]  /root/rtl8188eus/core/rtw_security.o
  CC [M]  /root/rtl8188eus/core/rtw_debug.o
  CC [M]  /root/rtl8188eus/core/rtw_io.o
  CC [M]  /root/rtl8188eus/core/rtw_ioctl_query.o
  CC [M]  /root/rtl8188eus/core/rtw_ioctl_set.o
  CC [M]  /root/rtl8188eus/core/rtw_ieee80211.o
  CC [M]  /root/rtl8188eus/core/rtw_mlme.o
  CC [M]  /root/rtl8188eus/core/rtw_mlme_ext.o
  CC [M]  /root/rtl8188eus/core/rtw_wlan_util.o
  CC [M]  /root/rtl8188eus/core/rtw_vht.o
  CC [M]  /root/rtl8188eus/core/rtw_pwrctrl.o
  CC [M]  /root/rtl8188eus/core/rtw_rf.o
  CC [M]  /root/rtl8188eus/core/rtw_recv.o
  CC [M]  /root/rtl8188eus/core/rtw_sta_mgt.o
  CC [M]  /root/rtl8188eus/core/rtw_ap.o
  CC [M]  /root/rtl8188eus/core/rtw_xmit.o
  CC [M]  /root/rtl8188eus/core/rtw_p2p.o
  CC [M]  /root/rtl8188eus/core/rtw_tdls.o
  CC [M]  /root/rtl8188eus/core/rtw_br_ext.o
/root/rtl8188eus/core/rtw_br_ext.c: In function ‘nat25_db_handle’:
/root/rtl8188eus/core/rtw_br_ext.c:1447:28: error: implicit declaration of function ‘csum_ipv6_magic’ [-Werror=implicit-function-declaration]
         hdr->icmp6_cksum = csum_ipv6_magic(&iph->saddr, &iph->daddr,
                            ^~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
scripts/Makefile.build:312: recipe for target '/root/rtl8188eus/core/rtw_br_ext.o' failed
make[2]: *** [/root/rtl8188eus/core/rtw_br_ext.o] Error 1
Makefile:1571: recipe for target '_module_/root/rtl8188eus' failed
make[1]: *** [_module_/root/rtl8188eus] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.17.11-sunxi64'
Makefile:1367: recipe for target 'modules' failed
make: *** [modules] Error 2
```

Any idea?

quickreflex, 2018-08-07:
The error is solved and fixed at 3fc2b04fd5cc8e3dbb74146f2f19887dfe71be3d commit.

EvilOlaf, 2018-08-07:
Wow, that was quick. Thanks. I'll give it a shot ASAP.

EvilOlaf, 2018-08-07:
Compiling flawless now.
Did not help me though has it seem not to support RTL8188FTV chip (which of course was not expected, just gave it a try).

closed 2018-08-07 by EvilOlaf