quickreflex/rtl8188eus #9: ~/Desktop/rtl8188eus# iw dev wlan0 set type monitor command failed: No such device (-19)

saikumar41, 2018-08-11:


tujimeaj, 2019-01-16:
still no solution ah?

k0ur0x, 2019-01-17:
I think you should see your device name with `iwconfig` command. For example, in my case, it's wlp0s20u2.

tujimeaj, 2019-01-17:
@k0ur0x it was not that case bro. my iwconfig showed wlan0...
@saikumar41, i did it. I deleted the previous build, git clone the kimocoder/rtl8188eus.git(cuz it already contained the fix with the line EXTRA_CFLAGS += -Wno-incompatible-pointer-types on Makefile), then i updated also os_intfs.c and wifi.h with the lines specified by luka-j https://github.com/quickreflex/rtl8188eus/pull/12. Then on issuing the command iw dev wlan0 set type monitor u should see no more errors, when verifyingthe iwconfig u should see auto mode.Hope this works also for you.

hazeuss, 2019-05-30:
@tujimeaj do you think you could help me please? im having the same issue but im not sure what you refer to when you mention the errors in #12  by @luka-j
thank you

GetRektBoy724, 2020-01-11:
is there any fix around?
:)

bakwc, 2021-08-24:
Had the same issue, installed this driver instead: https://github.com/aircrack-ng/rtl8188eus/