quickreflex/rtl8188eus #13: Minor Makefile changes to build in RPi 3b +

estcap2, 2018-11-22:
I managed to build the driver inside the RPi itself with this minor changes:

Changed the line `CONFIG_PLATFORM_I386_PC = y` to `CONFIG_PLATFORM_I386_PC = n`
Added the line `CONFIG_PLATFORM_ARM_RPI = y`

Added the following lines on the `ifeq($(CONFIG_PLATFORM_*` section
```
ifeq ($(CONFIG_PLATFORM_ARM_RPI), y)
EXTRA_CFLAGS += -DCONFIG_LITTLE_ENDIAN
ARCH ?= arm
CROSS_COMPILE ?=
KVER ?= $(shell uname -r)
KSRC := /lib/modules/$(KVER)/build
MODDESTDIR := /lib/modules/$(KVER)/kernel/drivers/net/wireless/
INSTALL_PREFIX :=
endif
```

I hope this might be useful for someone

Dragas, 2019-03-15:
This should be a pull request.

estcap2, 2019-03-15:
@Dragas I would suggest as such but this was not tested throughly since I got a WN722n V1 shortly after submitting this. I recall being able to build and install whith this modification, but afterwards having issues with injecting or some relevant features I wanted to use. I believe that capturing packets still worked well though.

Dragas, 2019-03-16:
Hmm, I agree. There are some issues with V3 as well. For example I can't switch channels and need to use legacy methods to setup monitor mode.

filipopo, 2023-12-23:
ummm after making these changes I'm getting this output... any ideas?
```
make ARCH=arm CROSS_COMPILE= -C /lib/modules/6.1.21-v8+/build M=/home/pi/temp/rtl8188eus  modules
make[1]: Entering directory '/usr/src/linux-headers-6.1.21-v8+'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: gcc (Debian 10.2.1-6) 10.2.1 20210110
  You are using:           gcc (Debian 12.2.0-14) 12.2.0
  CC [M]  /home/pi/temp/rtl8188eus/core/rtw_cmd.o
gcc: error: unrecognized argument in option ‘-mabi=apcs-gnu’
gcc: note: valid arguments to ‘-mabi=’ are: ilp32 lp64
gcc: error: unrecognized command-line option ‘-mapcs’
gcc: error: unrecognized command-line option ‘-mno-sched-prolog’
gcc: error: unrecognized command-line option ‘-msoft-float’
make[2]: *** [scripts/Makefile.build:250: /home/pi/temp/rtl8188eus/core/rtw_cmd.o] Error 1
make[1]: *** [Makefile:2012: /home/pi/temp/rtl8188eus] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.1.21-v8+'
make: *** [Makefile:1922: modules] Error 2
```

filipopo, 2023-12-23:
Well, this builds... https://github.com/ivanovborislav/rtl8188eu
So I took code from that project and replaced rtl8188eus/core/rtw_cmd.c:747:16 with "return 0;" and now it compiles...