quickreflex/rtl8188eus #15: Failed to build on 4.20

v1993, 2019-01-08:
I try to install this module onto linux kernel 4.20 and get following errors:

```
v@v-xubuntu:~/compile/rtl8188eus$ LANG=C sudo dkms build 8188eu/1.0

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make' all KVER=4.20.0-042000-generic.........(bad exit status: 2)
Error! Bad return status for module build on kernel: 4.20.0-042000-generic (x86_64)
Consult /var/lib/dkms/8188eu/1.0/build/make.log for more information.
v@v-xubuntu:~/compile/rtl8188eus$ cat /var/lib/dkms/8188eu/1.0/build/make.log
DKMS make.log for 8188eu-1.0 for kernel 4.20.0-042000-generic (x86_64)
Tue Jan  8 18:03:30 MSK 2019
"******************************************"
"NO SKRC,we will use default KSRC"
"******************************************"
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.20.0-042000-generic/build M=/var/lib/dkms/8188eu/1.0/build  modules
make[1]: Entering directory '/usr/src/linux-headers-4.20.0-042000-generic'
"******************************************"
"NO SKRC,we will use default KSRC"
"******************************************"
  CC [M]  /var/lib/dkms/8188eu/1.0/build/core/rtw_cmd.o
In file included from /var/lib/dkms/8188eu/1.0/build/include/drv_types.h:35,
                 from /var/lib/dkms/8188eu/1.0/build/core/rtw_cmd.c:22:
/var/lib/dkms/8188eu/1.0/build/include/wifi.h:988: warning: "IEEE80211_MAX_AMPDU_BUF" redefined
 #define IEEE80211_MAX_AMPDU_BUF 0x40

In file included from ./include/net/cfg80211.h:24,
                 from /var/lib/dkms/8188eu/1.0/build/include/osdep_service_linux.h:83,
                 from /var/lib/dkms/8188eu/1.0/build/include/osdep_service.h:45,
                 from /var/lib/dkms/8188eu/1.0/build/include/drv_types.h:32,
                 from /var/lib/dkms/8188eu/1.0/build/core/rtw_cmd.c:22:
./include/linux/ieee80211.h:1442: note: this is the location of the previous definition
 #define IEEE80211_MAX_AMPDU_BUF  0x100

  CC [M]  /var/lib/dkms/8188eu/1.0/build/core/rtw_security.o
In file included from /var/lib/dkms/8188eu/1.0/build/include/drv_types.h:35,
                 from /var/lib/dkms/8188eu/1.0/build/core/rtw_security.c:22:
/var/lib/dkms/8188eu/1.0/build/include/wifi.h:988: warning: "IEEE80211_MAX_AMPDU_BUF" redefined
 #define IEEE80211_MAX_AMPDU_BUF 0x40

In file included from ./include/net/cfg80211.h:24,
                 from /var/lib/dkms/8188eu/1.0/build/include/osdep_service_linux.h:83,
                 from /var/lib/dkms/8188eu/1.0/build/include/osdep_service.h:45,
                 from /var/lib/dkms/8188eu/1.0/build/include/drv_types.h:32,
                 from /var/lib/dkms/8188eu/1.0/build/core/rtw_security.c:22:
./include/linux/ieee80211.h:1442: note: this is the location of the previous definition
 #define IEEE80211_MAX_AMPDU_BUF  0x100

  CC [M]  /var/lib/dkms/8188eu/1.0/build/core/rtw_debug.o
In file included from /var/lib/dkms/8188eu/1.0/build/include/drv_types.h:35,
                 from /var/lib/dkms/8188eu/1.0/build/core/rtw_debug.c:22:
/var/lib/dkms/8188eu/1.0/build/include/wifi.h:988: warning: "IEEE80211_MAX_AMPDU_BUF" redefined
 #define IEEE80211_MAX_AMPDU_BUF 0x40

In file included from ./include/net/cfg80211.h:24,
                 from /var/lib/dkms/8188eu/1.0/build/include/osdep_service_linux.h:83,
                 from /var/lib/dkms/8188eu/1.0/build/include/osdep_service.h:45,
                 from /var/lib/dkms/8188eu/1.0/build/include/drv_types.h:32,
                 from /var/lib/dkms/8188eu/1.0/build/core/rtw_debug.c:22:
./include/linux/ieee80211.h:1442: note: this is the location of the previous definition
 #define IEEE80211_MAX_AMPDU_BUF  0x100

  CC [M]  /var/lib/dkms/8188eu/1.0/build/core/rtw_io.o
In file included from /var/lib/dkms/8188eu/1.0/build/include/drv_types.h:35,
                 from /var/lib/dkms/8188eu/1.0/build/core/rtw_io.c:52:
/var/lib/dkms/8188eu/1.0/build/include/wifi.h:988: warning: "IEEE80211_MAX_AMPDU_BUF" redefined
 #define IEEE80211_MAX_AMPDU_BUF 0x40
[... 319 lines cut ...]

  CC [M]  /var/lib/dkms/8188eu/1.0/build/os_dep/linux/os_intfs.o
In file included from /var/lib/dkms/8188eu/1.0/build/include/drv_types.h:35,
                 from /var/lib/dkms/8188eu/1.0/build/os_dep/linux/os_intfs.c:22:
/var/lib/dkms/8188eu/1.0/build/include/wifi.h:988: warning: "IEEE80211_MAX_AMPDU_BUF" redefined
 #define IEEE80211_MAX_AMPDU_BUF 0x40

In file included from ./include/net/cfg80211.h:24,
                 from /var/lib/dkms/8188eu/1.0/build/include/osdep_service_linux.h:83,
                 from /var/lib/dkms/8188eu/1.0/build/include/osdep_service.h:45,
                 from /var/lib/dkms/8188eu/1.0/build/include/drv_types.h:32,
                 from /var/lib/dkms/8188eu/1.0/build/os_dep/linux/os_intfs.c:22:
./include/linux/ieee80211.h:1442: note: this is the location of the previous definition
 #define IEEE80211_MAX_AMPDU_BUF  0x100

/var/lib/dkms/8188eu/1.0/build/os_dep/linux/os_intfs.c:742:22: error: initialization of 'u16 (*)(struct net_device *, struct sk_buff *, struct net_device *, u16 (*)(struct net_device *, struct sk_buff *, struct net_device *))' {aka 'short unsigned int (*)(struct net_device *, struct sk_buff *, struct net_device *, short unsigned int (*)(struct net_device *, struct sk_buff *, struct net_device *))'} from incompatible pointer type 'u16 (*)(struct net_device *, struct sk_buff *, void *, u16 (*)(struct net_device *, struct sk_buff *, struct net_device *))' {aka 'short unsigned int (*)(struct net_device *, struct sk_buff *, void *, short unsigned int (*)(struct net_device *, struct sk_buff *, struct net_device *))'} [-Werror=incompatible-pointer-types]
  .ndo_select_queue = rtw_select_queue,
                      ^~~~~~~~~~~~~~~~
/var/lib/dkms/8188eu/1.0/build/os_dep/linux/os_intfs.c:742:22: note: (near initialization for 'rtw_netdev_ops.ndo_select_queue')
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:292: /var/lib/dkms/8188eu/1.0/build/os_dep/linux/os_intfs.o] Error 1
make[1]: *** [Makefile:1563: _module_/var/lib/dkms/8188eu/1.0/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.20.0-042000-generic'
make: *** [Makefile:1367: modules] Error 2
```

codito, 2019-01-19:
https://github.com/quickreflex/rtl8188eus/pull/12 has a fix

quickreflex, 2019-03-22:
The fix is done.