quickreflex/rtl8188eus #17: Compiling Issue

jairook, 2019-01-29:
`[jaiagg@jai-pc rtl8188eus]$ make all
"******************************************"
"NO SKRC,we will use default KSRC"
"******************************************"
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.19.16-1-MANJARO/build M=/run/media/jaiagg/Personal/Work/new_drivers/rtl8188eus  modules
make[1]: Entering directory '/usr/lib/modules/4.19.16-1-MANJARO/build'
"******************************************"
"NO SKRC,we will use default KSRC"
"******************************************"
  CC [M]  /run/media/jaiagg/Personal/Work/new_drivers/rtl8188eus/os_dep/linux/os_intfs.o
In file included from /run/media/jaiagg/Personal/Work/new_drivers/rtl8188eus/include/drv_types.h:35,
                 from /run/media/jaiagg/Personal/Work/new_drivers/rtl8188eus/os_dep/linux/os_intfs.c:22:
/run/media/jaiagg/Personal/Work/new_drivers/rtl8188eus/include/wifi.h:988: warning: "IEEE80211_MAX_AMPDU_BUF" redefined
 #define IEEE80211_MAX_AMPDU_BUF 0x40
 
In file included from ./include/net/cfg80211.h:24,
                 from /run/media/jaiagg/Personal/Work/new_drivers/rtl8188eus/include/osdep_service_linux.h:83,
                 from /run/media/jaiagg/Personal/Work/new_drivers/rtl8188eus/include/osdep_service.h:45,
                 from /run/media/jaiagg/Personal/Work/new_drivers/rtl8188eus/include/drv_types.h:32,
                 from /run/media/jaiagg/Personal/Work/new_drivers/rtl8188eus/os_dep/linux/os_intfs.c:22:
./include/linux/ieee80211.h:1442: note: this is the location of the previous definition
 #define IEEE80211_MAX_AMPDU_BUF  0x100
 
/run/media/jaiagg/Personal/Work/new_drivers/rtl8188eus/os_dep/linux/os_intfs.c:742:22: error: initialization of ‘u16 (*)(struct net_device *, struct sk_buff *, struct net_device *, u16 (*)(struct net_device *, struct sk_buff *, struct net_device *))’ {aka ‘short unsigned int (*)(struct net_device *, struct sk_buff *, struct net_device *, short unsigned int (*)(struct net_device *, struct sk_buff *, struct net_device *))’} from incompatible pointer type ‘u16 (*)(struct net_device *, struct sk_buff *, void *, u16 (*)(struct net_device *, struct sk_buff *, struct net_device *))’ {aka ‘short unsigned int (*)(struct net_device *, struct sk_buff *, void *, short unsigned int (*)(struct net_device *, struct sk_buff *, struct net_device *))’} [-Werror=incompatible-pointer-types]
  .ndo_select_queue = rtw_select_queue,
                      ^~~~~~~~~~~~~~~~
/run/media/jaiagg/Personal/Work/new_drivers/rtl8188eus/os_dep/linux/os_intfs.c:742:22: note: (near initialization for ‘rtw_netdev_ops.ndo_select_queue’)
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:304: /run/media/jaiagg/Personal/Work/new_drivers/rtl8188eus/os_dep/linux/os_intfs.o] Error 1
make[1]: *** [Makefile:1520: _module_/run/media/jaiagg/Personal/Work/new_drivers/rtl8188eus] Error 2
make[1]: Leaving directory '/usr/lib/modules/4.19.16-1-MANJARO/build'
make: *** [Makefile:1367: modules] Error 2
`
Using Arch Linux distro 'Manjaro'.. Getting above error, any ideas?

dannyyy-jimenez, 2019-01-30:
First time answering a question here so I might not do it to convention. 

In this repo: https://github.com/smlinux/rtl8723de/issues/27

Someone had the same issue and a temporary fix was found, which also worked for me. 
Im running 4.19 Kali, so this outta fix your issue too. 

Mentioned by user @Boogie3D
"Just a temporary workaround, but you can still compile and install the driver if you add the following line to the top of the Makefile:

USER_EXTRA_CFLAGS=-Wno-error=incompatible-pointer-types"

Hope that helps

jairook, 2019-01-30:
@dannyyy-jimenez , Thanks for the reference and help.

So, the permanent solution would come out in a new version of Kernel, or kernel update, any idea?

Mattwmaster58, 2022-08-11:
@jairook, the permanent solution would be to update the method to match kernal api call signatures I believe