quickreflex/rtl8188eus #18: New version available from Realtek: v5.3.9

kimocoder, 2019-01-30:
Hi there!

There is a new version available for this driver.
Changelog is found [here](https://github.com/kimocoder/rtl8188eus/blob/v5.3.9/RTL8188EUS%20ReleaseNotes.pdf)

**Howto get it from Realtek?**
Ftp Server : ftp3.realtek.com
Account:RTL8188E Password: M6b3Er (8188EUS/ETV/8189ES)

minhducsun2002, 2019-02-12:
Confirmed to work with [some editing](https://github.com/kimocoder/rtl8188eus/pull/1).

The following things are tested :
- Monitor mode (you have to get your hands dirty a little with `sudo ifconfig wlan1 down` and `sudo iwconfig wlan1 mode monitor`) : `airodump-ng` works. `aireplay-ng` successfully deauthenticates me.
- Injection test : well, readers, decide it:
```
root@magister-mentium [0] /usr/src/rtl8188eus # aireplay-ng -9 -i wlan0mon wlan1   

19:33:23  Trying broadcast probe requests...
19:33:23  Injection is working!
19:33:25  Found 3 APs

19:33:25  Trying directed probe requests...
19:33:25  <MAC, censored> - channel: 11 - <network name, censored>
19:33:25  Ping (min/avg/max): 2.006ms/8.028ms/19.423ms Power: -67.60
19:33:25  30/30: 100%

19:33:25  <MAC, censored> - channel: 10 - <network name, censored>
19:33:26  Ping (min/avg/max): 2.694ms/17.707ms/36.812ms Power: -76.30
19:33:26  30/30: 100%

19:33:26  <MAC, censored> - channel: 9 - <network name, censored>
19:33:32  Ping (min/avg/max): 9.700ms/9.700ms/9.700ms Power: -69.00
19:33:32   1/30:   3%

19:33:32  Trying card-to-card injection...
19:33:36  Attack -0:           Failed
19:33:40  Attack -1 (open):    Failed
19:33:41  Attack -1 (psk):     OK
19:33:46  Attack -2/-3/-4/-6:  Failed
19:33:50  Attack -5/-7:        Failed
```

  _It's not the problem with the APs if you ask; my PCIe AR9285 perfectly passes._

kimocoder, 2019-02-12:
Don't worry about the errors, I created them when adding adapter support yesterday.

Anyhow, clean source is available from ftp3.realtek.com.

And I'll add support to aircrack-ng (in airmon-ng) within a few days too, so you don't need to set it manually.

minhducsun2002, 2019-02-12:
> Don't worry about the errors, I created them when adding adapter support yesterday.

Not that much errors, but mate, can you make your driver speak less? :) It's bloating my `dmesg` output.

> And I'll add support to aircrack-ng (in airmon-ng) within a few days too, so you don't need to set it manually.

Just testing things a little in case someone hesitates.

Maybe I'll give it a try on Arch's official 4.20 kernel tomorrow.

kimocoder, 2019-02-12:
Yes, I have lots to do, so I need the debug for now, but I'll remove the most spamming one this evening.

minhducsun2002, 2019-02-12:
> this evening.

I'm at GMT+7, which means, at the time of writing this it would already been night for me though :| Anyway, looking forward for your intervention to make the driver less talkative. :)

kimocoder, 2019-02-12:
I pushed some fixes, I'll do some more important ones tomorrow too

quickreflex, 2019-03-22:
Wonderful! Thanks for the new version. I will try it and I can open another branch.

quickreflex, 2019-03-24:
I'm uploading the latest patches for this new version. Included are updates in the Linux kernel at 4.19, 4.20, 5.0 and 5.1.
[v5.3.9_28540.tar.gz](https://github.com/quickreflex/rtl8188eus/files/3000387/v5.3.9_28540.tar.gz)

bobby69-24, 2020-06-23:
plz help me ! tell me what kernel you use you using im totally tired of install too many kernels and trying  [im getting errors in latest kernel and in old kernel installs sucessfully with warnings but wifi wont work]