quickreflex/rtl8188eus #28: Build fix for linux 5.2.x

codito, 2019-08-17:
```diff
diff --git a/os_dep/linux/os_intfs.c b/os_dep/linux/os_intfs.c
index b183a09..478f384 100644
--- a/os_dep/linux/os_intfs.c
+++ b/os_dep/linux/os_intfs.c
@@ -1099,7 +1099,10 @@ unsigned int rtw_classify8021d(struct sk_buff *skb)
 }
 
 
-#if (LINUX_VERSION_CODE>=KERNEL_VERSION(4,19,0))
+#if (LINUX_VERSION_CODE>=KERNEL_VERSION(5,2,0))
+static u16 rtw_select_queue(struct net_device *dev, struct sk_buff *skb,
+		struct net_device *sb_dev)
+#elif (LINUX_VERSION_CODE>=KERNEL_VERSION(4,19,0))
 static u16 rtw_select_queue(struct net_device *dev, struct sk_buff *skb,
 		struct net_device *sb_dev,
 		select_queue_fallback_t fallback)
```

Tested with `5.2.8` on Arch linux.

FlaccidNic, 2019-09-28:
I've a problem with rtl8188eus can't install it in kali linux  5.2.0 

Error! Bad return status for module build on kernel: 5.2.0-kali2-amd64 (x86_64)

SirGrifo, 2019-09-29:
same here, can you help?

Building module:
cleaning build area...
'make' all KVER=5.2.0-2parrot1-amd64...............(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.2.0-2parrot1-amd64 (x86_64)
Consult /var/lib/dkms/8188eu/1.0/build/make.log for more information.

codito, 2019-09-30:
Does the content of `/var/lib/dkms/8188eu/1.0/build/make.log` provide any hints?

aleek, 2019-10-15:
I confirm this patch fixes compilation in gentoo:
`linux 5.2.0-pf8
gcc version 8.3.0 (Gentoo 8.3.0-r1 p1.1)
binutils x86_64-pc-linux-gnu-2.32`

HabeebUllah555, 2019-10-19:
> I confirm this patch fixes compilation in gentoo:
> `linux 5.2.0-pf8 gcc version 8.3.0 (Gentoo 8.3.0-r1 p1.1) binutils x86_64-pc-linux-gnu-2.32`

Cloned a day before but make all is stuck up and giving error, already started a new issue, but no response yet.