quickreflex/rtl8188eus #30: i made it , monitor mode working on rtl8188eus ( tl-wn722n v3)

hameddanesh, 2019-09-10:
hey guys i finally made it work thanks to tips on quickreflex and driver version 5.3.9 on kimocoder
to make monitoring work on kali linux rolling 2019 frist you have to use 4.19 kernell witch is kali rolling 2019.a1 orginal kernel ( or install a lower kernal ( a linux image )
then go to :
https://github.com/kimocoder/rtl8188eus/tree/v5.3.9
download the driver , extract it in a folder , right click in folder and choose Open in termianl
and then enter these commands :
make clean
make all
cp realtek_blacklist.conf /lib/modprobe.d/
make install
reboot

to activate monitor mode : 
ifconfig <wlan name> down
iwconfig <wlan name> mode monitor
ifconfig <wlan name> up

i worked 3 day to find this , i hope this way will work for you too.
but even if not this is at least answer for a question most of us have : yes tl-wn722n v3 ( rtl8188eus chipset ) do work on monitoring mode in kali linux.

closed 2019-09-10 by hameddanesh

Sqxxe, 2020-02-11:
Hey can I get your discord so you can help me out since you've most likely went through this.

wallyradburn, 2020-02-19:
i get the following error in terminal when running the cp command :
cp: missing destination file operand after realtek_blacklist.conf /lib/modprobe.d/

can anyone help me with this please

hameddanesh, 2020-02-24:
> i get the following error in terminal when running the cp command :
> cp: missing destination file operand after realtek_blacklist.conf /lib/modprobe.d/
> 
> can anyone help me with this please

well did you check the kernel version ? 🤔

php717, 2020-04-08:
TP-LINK official have a good full manual how to do driver install 
https://www.tp-link.com/ae/support/download/tl-wn722n/v3/#Driver
going to try now..
shame that nethunter 2020 edition dont have drivers preloaded

benkio, 2020-04-23:
A little comment, to compile the driver https://github.com/kimocoder/rtl8188eus/tree/v5.3.9
Use gcc-7!! I have gcc 7.5

now I continue :crossed_fingers:

Chris-Zoumbos, 2020-05-09:
Is there any way to use the beta drivers on the latest Kali Linux?

UchihaItachi1416, 2020-05-30:
Can someone make a tutorial how to do everything? I'm new to this and can't even understand a thing!

TuserAhammed, 2020-06-18:
Tell me sir,,, Tp link wn722n V3 is working in nethunte??   And package injection  is working or not??? And  can i used this v3 adapter in fluxio???

Zura001, 2020-07-07:
> i get the following error in terminal when running the cp command :
> cp: missing destination file operand after realtek_blacklist.conf /lib/modprobe.d/
> 
> can anyone help me with this please

Delete the "/" after modprob.d

namanh2710, 2020-07-20:
tksssss so much it worked !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!

farhadankhi, 2021-06-16:
I tried my best to install the driver for my kali Linux but I couldn't. Can anyone help me pls?

GetRektBoy724, 2021-06-17:
if you're using Kali Linux or you have kali linux's repository, just install `realtek-rtl8188eus-dkms` with apt

farhadankhi, 2021-06-17:
I am using kali Linux with VM Virtual Box...

farhadankhi, 2021-06-17:
> if you're using Kali Linux or you have kali linux's repository, just install `realtek-rtl8188eus-dkms` with apt

what do you mean by "with apt"?