quickreflex/rtl8188eus #31: hey guys i made it , rtl8188eus do monitoring on kali linux rolling

hameddanesh, 2019-09-10:
hey guys i finally made it work thanks to tips on quickreflex and driver version 5.3.9 on kimocoder
to make monitoring work on kali linux rolling 2019 frist you have to use 4.19 kernell which is kali rolling 2019.a1 orginal kernel ( or install a lower kernal ( a linux image )
then go to :
https://github.com/kimocoder/rtl8188eus/tree/v5.3.9
download the driver , extract it in a folder , right click in folder and choose Open in termianl
and then enter these commands :
make clean
make all
cp realtek_blacklist.conf /lib/modprobe.d/
make install
reboot

to activate monitor mode : 
ifconfig <wlan name> down
iwconfig <wlan name> mode monitor
ifconfig <wlan name> up

i worked 3 day to find this , i hope this way will work for you too.
but even if not this is at least answer for a question most of us have : yes tl-wn722n v3 ( rtl8188eus chipset ) do work on monitoring mode in kali linux.

hameddanesh, 2019-09-11:
i just forgot to write , to use this every time on boot you gota go to grub advanced options ( if i is not being displayed hold shift button on system boot start )  and select the kernel version you want.

Reddymanjesh, 2019-09-13:
Hi @hameddanesh , Congo on your success. Can you help me with the way you did, please ?
I have v5 kali 2019 , tl-wn722n  wifi adapter v3. 

Reagrds,
Manjesh Reddy

kimocoder, 2019-09-13:
woot woot

hameddanesh, 2019-09-27:
@reddymanjesh hi , sure i can.
sorry for answering late , i was a bit busy.
for doing so , first check which versions of kernel you have. if you do not have a kernel lower than kimocoder's driver wont work on your system so you need to install another lower version of kernel so that you can run the driver on your system. ( you can fine terminal commands for it on google )
after you done this , 
A) on system power up hold shift , then on grub loader select advanced and there you can choose the kernel version you want ( the lower version you installed )
after the os loaded do as i explained above
dont forget each time you wana use kimo's driver you need to fallow step A to boot with comfortable kernel for driver

hameddanesh, 2019-09-27:
@kimocoder thank you for your good driver and help full tips.
i wanted to message you then and thank you but i thought you are busy and may not have time for these but i gave stars both to you and quickreflex to do so.
but i think now i can write here : thank you for being really helpfull.

bobby69-24, 2020-06-23:
it donot work for me install sucessfully with warnings and wifi wont work ...

bobby69-24, 2020-06-23:
and i installed kernel 4.19 plz help hameddanesh

closed 2021-11-15 by hameddanesh