quickreflex/rtl8188eus #35: make file error

Qurram555, 2019-10-18:
After 
/home/admin/rtl8188eus/os_dep/linux/os_intfs.c:1240:22: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
  .ndo_select_queue = rtw_select_queue,
                      ^~~~~~~~~~~~~~~~

StefanosAstaras, 2019-11-13:
Yup, same problem here. It looks like a field was removed from the ndo_select_queue signature in kernels >=5.2 and rtw_select_queue is no longer compatible. I naively tried to remove the same field from the latter, but it's not that simple (compiles but crashes).

iandol, 2020-01-09:
Same error on 5.3.0-26-generic #28-Ubuntu

cstayyab, 2020-01-29:
Adding 
`EXTRA_CFLAGS += -Wno-incompatible-pointer-types`
to Makefile solves the problem

marcosriso, 2020-02-27:
The change above in makefile works, but now I have this problem when running make all:

error: macro "access_ok" passed 3 arguments, but takes just 2
  if (!access_ok(VERIFY_READ, priv_cmd.buf, priv_cmd.total_len)) {

error: ‘access_ok’ undeclared (first use in this function)
  if (!access_ok(VERIFY_READ, priv_cmd.buf, priv_cmd.total_len)) {