quickreflex/rtl8188eus #36: Error installing on kali linux 2019.3 (kernal 5.3.0-kali2-amd64)

foundation-university, 2019-11-16:
I am try to install this driver on kali linux 2019.3 (kernal 5.3.0-kali2-amd64). I followed the instruction and when i execute 
```
make all
```
commands i and getting this error
```
/bin/sh: 1: bc: not found
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.3.0-kali2-amd64/build M=/root/Desktop/rtl8188eus_1  modules
make[1]: Entering directory '/usr/src/linux-headers-5.3.0-kali2-amd64'
/bin/sh: 1: bc: not found
  CC [M]  /root/Desktop/rtl8188eus_1/core/rtw_cmd.o
  CC [M]  /root/Desktop/rtl8188eus_1/core/rtw_security.o
  CC [M]  /root/Desktop/rtl8188eus_1/core/rtw_debug.o
  CC [M]  /root/Desktop/rtl8188eus_1/core/rtw_io.o
  CC [M]  /root/Desktop/rtl8188eus_1/core/rtw_ioctl_query.o
  CC [M]  /root/Desktop/rtl8188eus_1/core/rtw_ioctl_set.o
  CC [M]  /root/Desktop/rtl8188eus_1/core/rtw_ieee80211.o
  CC [M]  /root/Desktop/rtl8188eus_1/core/rtw_mlme.o
  CC [M]  /root/Desktop/rtl8188eus_1/core/rtw_mlme_ext.o
/root/Desktop/rtl8188eus_1/core/rtw_mlme_ext.c: In function ‘mgt_dispatcher’:
/root/Desktop/rtl8188eus_1/core/rtw_mlme_ext.c:1833:17: warning: this statement may fall through [-Wimplicit-fallthrough=]
 1833 |    ptable->func = &OnAuthClient;
      |    ~~~~~~~~~~~~~^~~~~~~~~~~~~~~
/root/Desktop/rtl8188eus_1/core/rtw_mlme_ext.c:1835:2: note: here
 1835 |  case WIFI_ASSOCREQ:
      |  ^~~~
  CC [M]  /root/Desktop/rtl8188eus_1/core/rtw_mi.o
  CC [M]  /root/Desktop/rtl8188eus_1/core/rtw_wlan_util.o
  CC [M]  /root/Desktop/rtl8188eus_1/core/rtw_vht.o
  CC [M]  /root/Desktop/rtl8188eus_1/core/rtw_pwrctrl.o
  CC [M]  /root/Desktop/rtl8188eus_1/core/rtw_rf.o
  CC [M]  /root/Desktop/rtl8188eus_1/core/rtw_recv.o
  CC [M]  /root/Desktop/rtl8188eus_1/core/rtw_sta_mgt.o
  CC [M]  /root/Desktop/rtl8188eus_1/core/rtw_ap.o
/root/Desktop/rtl8188eus_1/core/rtw_ap.c: In function ‘rtw_add_bcn_ie’:
/root/Desktop/rtl8188eus_1/core/rtw_ap.c:195:25: warning: ‘ielen’ may be used uninitialized in this function [-Wmaybe-uninitialized]
  195 |  if (p != NULL && ielen > 0) {
      |                   ~~~~~~^~~
  CC [M]  /root/Desktop/rtl8188eus_1/core/mesh/rtw_mesh.o
  CC [M]  /root/Desktop/rtl8188eus_1/core/mesh/rtw_mesh_pathtbl.o
  CC [M]  /root/Desktop/rtl8188eus_1/core/mesh/rtw_mesh_hwmp.o
  CC [M]  /root/Desktop/rtl8188eus_1/core/rtw_xmit.o
  CC [M]  /root/Desktop/rtl8188eus_1/core/rtw_p2p.o
  CC [M]  /root/Desktop/rtl8188eus_1/core/rtw_rson.o
  CC [M]  /root/Desktop/rtl8188eus_1/core/rtw_tdls.o
  CC [M]  /root/Desktop/rtl8188eus_1/core/rtw_br_ext.o
  CC [M]  /root/Desktop/rtl8188eus_1/core/rtw_iol.o
  CC [M]  /root/Desktop/rtl8188eus_1/core/rtw_sreset.o
  CC [M]  /root/Desktop/rtl8188eus_1/core/rtw_btcoex_wifionly.o
  CC [M]  /root/Desktop/rtl8188eus_1/core/rtw_btcoex.o
  CC [M]  /root/Desktop/rtl8188eus_1/core/rtw_beamforming.o
  CC [M]  /root/Desktop/rtl8188eus_1/core/rtw_odm.o
  CC [M]  /root/Desktop/rtl8188eus_1/core/rtw_rm.o
  CC [M]  /root/Desktop/rtl8188eus_1/core/rtw_rm_fsm.o
  CC [M]  /root/Desktop/rtl8188eus_1/core/efuse/rtw_efuse.o
  CC [M]  /root/Desktop/rtl8188eus_1/os_dep/osdep_service.o
  CC [M]  /root/Desktop/rtl8188eus_1/os_dep/linux/os_intfs.o
  CC [M]  /root/Desktop/rtl8188eus_1/os_dep/linux/usb_intf.o
  CC [M]  /root/Desktop/rtl8188eus_1/os_dep/linux/usb_ops_linux.o
  CC [M]  /root/Desktop/rtl8188eus_1/os_dep/linux/ioctl_linux.o
  CC [M]  /root/Desktop/rtl8188eus_1/os_dep/linux/xmit_linux.o
  CC [M]  /root/Desktop/rtl8188eus_1/os_dep/linux/mlme_linux.o
  CC [M]  /root/Desktop/rtl8188eus_1/os_dep/linux/recv_linux.o
  CC [M]  /root/Desktop/rtl8188eus_1/os_dep/linux/ioctl_cfg80211.o
/root/Desktop/rtl8188eus_1/os_dep/linux/ioctl_cfg80211.c: In function ‘rtw_cfg80211_indicate_sta_assoc’:
/root/Desktop/rtl8188eus_1/os_dep/linux/ioctl_cfg80211.c:4141:3: warning: ISO C90 forbids mixed declarations and code [-Wdeclaration-after-statement]
 4141 |   u8 ie_offset;
      |   ^~
/root/Desktop/rtl8188eus_1/os_dep/linux/ioctl_cfg80211.c: In function ‘cfg80211_rtw_change_iface’:
/root/Desktop/rtl8188eus_1/os_dep/linux/ioctl_cfg80211.c:2368:10: warning: this statement may fall through [-Wimplicit-fallthrough=]
[... 92 lines cut ...]
  CC [M]  /root/Desktop/rtl8188eus_1/hal/phydm/halrf/rtl8188e/halrf_8188e_ce.o
  CC [M]  /root/Desktop/rtl8188eus_1/hal/phydm/rtl8188e/phydm_regconfig8188e.o
  CC [M]  /root/Desktop/rtl8188eus_1/hal/phydm/rtl8188e/hal8188erateadaptive.o
  CC [M]  /root/Desktop/rtl8188eus_1/hal/phydm/rtl8188e/phydm_rtl8188e.o
  CC [M]  /root/Desktop/rtl8188eus_1/platform/platform_ops.o
  CC [M]  /root/Desktop/rtl8188eus_1/core/rtw_mp.o
  LD [M]  /root/Desktop/rtl8188eus_1/8188eu.o
  Building modules, stage 2.
/bin/sh: 1: bc: not found
  MODPOST 1 modules
  CC      /root/Desktop/rtl8188eus_1/8188eu.mod.o
  LD [M]  /root/Desktop/rtl8188eus_1/8188eu.ko
make[1]: Leaving directory '/usr/src/linux-headers-5.3.0-kali2-amd64'
```
```
make install
```
commands i and getting this error
```
/bin/sh: 1: bc: not found
install -p -m 644 8188eu.ko  /lib/modules/5.3.0-kali2-amd64/kernel/drivers/net/wireless/
/sbin/depmod -a 5.3.0-kali2-amd64
```

Kindly guide main. how to intall the drivers

kimocoder, 2019-11-16:
apt-get install bc

foundation-university, 2019-11-17:
i execute this command `apt-get install bc` and still two errors and response of `make all`
```
root@kali:~/Desktop/rtl8188eus# make all
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.3.0-kali2-amd64/build M=/root/Desktop/rtl8188eus  modules
make[1]: Entering directory '/usr/src/linux-headers-5.3.0-kali2-amd64'
  CC [M]  /root/Desktop/rtl8188eus/core/rtw_cmd.o
  CC [M]  /root/Desktop/rtl8188eus/core/rtw_security.o
  CC [M]  /root/Desktop/rtl8188eus/core/rtw_debug.o
  CC [M]  /root/Desktop/rtl8188eus/core/rtw_io.o
  CC [M]  /root/Desktop/rtl8188eus/core/rtw_ioctl_query.o
  CC [M]  /root/Desktop/rtl8188eus/core/rtw_ioctl_set.o
  CC [M]  /root/Desktop/rtl8188eus/core/rtw_ieee80211.o
  CC [M]  /root/Desktop/rtl8188eus/core/rtw_mlme.o
  CC [M]  /root/Desktop/rtl8188eus/core/rtw_mlme_ext.o
/root/Desktop/rtl8188eus/core/rtw_mlme_ext.c: In function ‘mgt_dispatcher’:
/root/Desktop/rtl8188eus/core/rtw_mlme_ext.c:1441:6: warning: this statement may fall through [-Wimplicit-fallthrough=]
 1441 |   if (check_fwstate(pmlmepriv, WIFI_AP_STATE) == _TRUE)
      |      ^
/root/Desktop/rtl8188eus/core/rtw_mlme_ext.c:1446:2: note: here
 1446 |  case WIFI_ASSOCREQ:
      |  ^~~~
  CC [M]  /root/Desktop/rtl8188eus/core/rtw_mi.o
  CC [M]  /root/Desktop/rtl8188eus/core/rtw_wlan_util.o
  CC [M]  /root/Desktop/rtl8188eus/core/rtw_vht.o
  CC [M]  /root/Desktop/rtl8188eus/core/rtw_pwrctrl.o
  CC [M]  /root/Desktop/rtl8188eus/core/rtw_rf.o
  CC [M]  /root/Desktop/rtl8188eus/core/rtw_recv.o
  CC [M]  /root/Desktop/rtl8188eus/core/rtw_sta_mgt.o
  CC [M]  /root/Desktop/rtl8188eus/core/rtw_ap.o
/root/Desktop/rtl8188eus/core/rtw_ap.c: In function ‘rtw_add_bcn_ie’:
/root/Desktop/rtl8188eus/core/rtw_ap.c:193:9: warning: ‘ielen’ may be used uninitialized in this function [-Wmaybe-uninitialized]
  193 |   ielen += 2;
      |   ~~~~~~^~~~
  CC [M]  /root/Desktop/rtl8188eus/core/rtw_xmit.o
  CC [M]  /root/Desktop/rtl8188eus/core/rtw_p2p.o
  CC [M]  /root/Desktop/rtl8188eus/core/rtw_tdls.o
  CC [M]  /root/Desktop/rtl8188eus/core/rtw_br_ext.o
  CC [M]  /root/Desktop/rtl8188eus/core/rtw_iol.o
  CC [M]  /root/Desktop/rtl8188eus/core/rtw_sreset.o
  CC [M]  /root/Desktop/rtl8188eus/core/rtw_btcoex_wifionly.o
  CC [M]  /root/Desktop/rtl8188eus/core/rtw_btcoex.o
  CC [M]  /root/Desktop/rtl8188eus/core/rtw_beamforming.o
  CC [M]  /root/Desktop/rtl8188eus/core/rtw_odm.o
  CC [M]  /root/Desktop/rtl8188eus/core/efuse/rtw_efuse.o
  CC [M]  /root/Desktop/rtl8188eus/os_dep/osdep_service.o
  CC [M]  /root/Desktop/rtl8188eus/os_dep/linux/os_intfs.o
/root/Desktop/rtl8188eus/os_dep/linux/os_intfs.c:1240:22: error: initialization of ‘u16 (*)(struct net_device *, struct sk_buff *, struct net_device *)’ {aka ‘short unsigned int (*)(struct net_device *, struct sk_buff *, struct net_device *)’} from incompatible pointer type ‘u16 (*)(struct net_device *, struct sk_buff *, struct net_device *, u16 (*)(struct net_device *, struct sk_buff *, struct net_device *))’ {aka ‘short unsigned int (*)(struct net_device *, struct sk_buff *, struct net_device *, short unsigned int (*)(struct net_device *, struct sk_buff *, struct net_device *))’} [-Werror=incompatible-pointer-types]
 1240 |  .ndo_select_queue = rtw_select_queue,
      |                      ^~~~~~~~~~~~~~~~
/root/Desktop/rtl8188eus/os_dep/linux/os_intfs.c:1240:22: note: (near initialization for ‘rtw_netdev_ops.ndo_select_queue’)
cc1: some warnings being treated as errors
make[3]: *** [/usr/src/linux-headers-5.3.0-kali2-common/scripts/Makefile.build:286: /root/Desktop/rtl8188eus/os_dep/linux/os_intfs.o] Error 1
make[2]: *** [/usr/src/linux-headers-5.3.0-kali2-common/Makefile:1639: _module_/root/Desktop/rtl8188eus] Error 2
make[1]: *** [/usr/src/linux-headers-5.3.0-kali2-common/Makefile:179: sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.3.0-kali2-amd64'
make: *** [Makefile:1911: modules] Error 2
```

kimocoder, 2019-11-17:
I'll add a patch for it, just hold on

kimocoder, 2019-11-17:
In the rtl8188eus folder do the below..

make clean
wget https://github.com/quickreflex/rtl8188eus/pull/37/commits/0342a5197467a94d60f17cd0c2bf2a7644f51f2c.patch

patch -p1 < *. patch
rm -rf *.patch

make && make install

reboot.. loaded

LittleEumel, 2019-11-19:
I tried it, using the patch. 
The error persists.

/root/rtl8188eus/os_dep/linux/os_intfs.c:1240:22: error: initialization of ‘u16 (*)(struct net_device *, struct sk_buff *, struct net_device *)’ {aka ‘short unsigned int (*)(struct net_device *, struct sk_buff *, struct net_device *)’} from incompatible pointer type ‘u16 (*)(struct net_device *, struct sk_buff *, struct net_device *, u16 (*)(struct net_device *, struct sk_buff *, struct net_device *))’ {aka ‘short unsigned int (*)(struct net_device *, struct sk_buff *, struct net_device *, short unsigned int (*)(struct net_device *, struct sk_buff *, struct net_device *))’} [-Werror=incompatible-pointer-types]
 1240 |  .ndo_select_queue = rtw_select_queue,
      |                      ^~~~~~~~~~~~~~~~
/root/rtl8188eus/os_dep/linux/os_intfs.c:1240:22: note: (near initialization for ‘rtw_netdev_ops.ndo_select_queue’)

Comment out line 1239 - 1241 let me compile, but driver didnt work.

foundation-university, 2019-11-20:
I also tried the patch but not successful.
```
root@kali:~/Desktop/rtl8188eus# patch -p1 < *.patch
patching file os_dep/linux/os_intfs.c
root@kali:~/Desktop/rtl8188eus# rm -rf *.patch
root@kali:~/Desktop/rtl8188eus# make && make install
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.3.0-kali2-amd64/build M=/root/Desktop/rtl8188eus  modules
make[1]: Entering directory '/usr/src/linux-headers-5.3.0-kali2-amd64'
  CC [M]  /root/Desktop/rtl8188eus/core/rtw_cmd.o
  CC [M]  /root/Desktop/rtl8188eus/core/rtw_security.o
  CC [M]  /root/Desktop/rtl8188eus/core/rtw_debug.o
  CC [M]  /root/Desktop/rtl8188eus/core/rtw_io.o
  CC [M]  /root/Desktop/rtl8188eus/core/rtw_ioctl_query.o
  CC [M]  /root/Desktop/rtl8188eus/core/rtw_ioctl_set.o
  CC [M]  /root/Desktop/rtl8188eus/core/rtw_ieee80211.o
  CC [M]  /root/Desktop/rtl8188eus/core/rtw_mlme.o
  CC [M]  /root/Desktop/rtl8188eus/core/rtw_mlme_ext.o
/root/Desktop/rtl8188eus/core/rtw_mlme_ext.c: In function ‘mgt_dispatcher’:
/root/Desktop/rtl8188eus/core/rtw_mlme_ext.c:1441:6: warning: this statement may fall through [-Wimplicit-fallthrough=]
 1441 |   if (check_fwstate(pmlmepriv, WIFI_AP_STATE) == _TRUE)
      |      ^
/root/Desktop/rtl8188eus/core/rtw_mlme_ext.c:1446:2: note: here
 1446 |  case WIFI_ASSOCREQ:
      |  ^~~~
  CC [M]  /root/Desktop/rtl8188eus/core/rtw_mi.o
  CC [M]  /root/Desktop/rtl8188eus/core/rtw_wlan_util.o
  CC [M]  /root/Desktop/rtl8188eus/core/rtw_vht.o
  CC [M]  /root/Desktop/rtl8188eus/core/rtw_pwrctrl.o
  CC [M]  /root/Desktop/rtl8188eus/core/rtw_rf.o
  CC [M]  /root/Desktop/rtl8188eus/core/rtw_recv.o
  CC [M]  /root/Desktop/rtl8188eus/core/rtw_sta_mgt.o
  CC [M]  /root/Desktop/rtl8188eus/core/rtw_ap.o
/root/Desktop/rtl8188eus/core/rtw_ap.c: In function ‘rtw_add_bcn_ie’:
/root/Desktop/rtl8188eus/core/rtw_ap.c:193:9: warning: ‘ielen’ may be used uninitialized in this function [-Wmaybe-uninitialized]
  193 |   ielen += 2;
      |   ~~~~~~^~~~
  CC [M]  /root/Desktop/rtl8188eus/core/rtw_xmit.o
  CC [M]  /root/Desktop/rtl8188eus/core/rtw_p2p.o
  CC [M]  /root/Desktop/rtl8188eus/core/rtw_tdls.o
  CC [M]  /root/Desktop/rtl8188eus/core/rtw_br_ext.o
  CC [M]  /root/Desktop/rtl8188eus/core/rtw_iol.o
  CC [M]  /root/Desktop/rtl8188eus/core/rtw_sreset.o
  CC [M]  /root/Desktop/rtl8188eus/core/rtw_btcoex_wifionly.o
  CC [M]  /root/Desktop/rtl8188eus/core/rtw_btcoex.o
  CC [M]  /root/Desktop/rtl8188eus/core/rtw_beamforming.o
  CC [M]  /root/Desktop/rtl8188eus/core/rtw_odm.o
  CC [M]  /root/Desktop/rtl8188eus/core/efuse/rtw_efuse.o
  CC [M]  /root/Desktop/rtl8188eus/os_dep/osdep_service.o
  CC [M]  /root/Desktop/rtl8188eus/os_dep/linux/os_intfs.o
/root/Desktop/rtl8188eus/os_dep/linux/os_intfs.c:1240:22: error: initialization of ‘u16 (*)(struct net_device *, struct sk_buff *, struct net_device *)’ {aka ‘short unsigned int (*)(struct net_device *, struct sk_buff *, struct net_device *)’} from incompatible pointer type ‘u16 (*)(struct net_device *, struct sk_buff *, struct net_device *, u16 (*)(struct net_device *, struct sk_buff *, struct net_device *))’ {aka ‘short unsigned int (*)(struct net_device *, struct sk_buff *, struct net_device *, short unsigned int (*)(struct net_device *, struct sk_buff *, struct net_device *))’} [-Werror=incompatible-pointer-types]
 1240 |  .ndo_select_queue = rtw_select_queue,
      |                      ^~~~~~~~~~~~~~~~
/root/Desktop/rtl8188eus/os_dep/linux/os_intfs.c:1240:22: note: (near initialization for ‘rtw_netdev_ops.ndo_select_queue’)
cc1: some warnings being treated as errors
make[3]: *** [/usr/src/linux-headers-5.3.0-kali2-common/scripts/Makefile.build:286: /root/Desktop/rtl8188eus/os_dep/linux/os_intfs.o] Error 1
make[2]: *** [/usr/src/linux-headers-5.3.0-kali2-common/Makefile:1639: _module_/root/Desktop/rtl8188eus] Error 2
make[1]: *** [/usr/src/linux-headers-5.3.0-kali2-common/Makefile:179: sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.3.0-kali2-amd64'
make: *** [Makefile:1911: modules] Error 2
```

foundation-university, 2019-11-22:
@kimocoder have you find any solution? I am waiting for your answer.

thewhiteh4t, 2019-11-23:
@kimocoder tried this patch method same error

gu2rks, 2019-11-29:
Got same error on kali 2019.4

Qaisser, 2019-12-31:
going to this 
/lib/modules/5.3.0-kali3-amd64/kernel/drivers/net/wireless
and do
rmmod 8188eu
insmod8188eu.ko

will don

LuD1161, 2020-05-13:
@Qaisser  @foundation-university  @gu2rks  @thewhiteh4t 

Use the pull request : https://github.com/quickreflex/rtl8188eus/pull/40
Probably you would be having kernel >= 5.1 ( like myself )

I have run this, works fine .
[image]

Also there's a space in 
> patch -p1 < *. patch

it should be `patch -p1 < *.patch` 

> In the rtl8188eus folder do the below..
> 
> make clean
> wget https://github.com/quickreflex/rtl8188eus/pull/37/commits/0342a5197467a94d60f17cd0c2bf2a7644f51f2c.patch
> 
> patch -p1 < *. patch
> rm -rf *.patch
> 
> make && make install
> 
> reboot.. loaded

Qaisser, 2020-05-14:
give me your number

LuD1161, 2020-05-14:
Well it got built correctly but still doesn't work.
I rebooted but still didn't work.

bobby69-24, 2020-06-11:
plz give me solution i too have same issue in kali vmware