quickreflex/rtl8188eus #48: rtl8188eus/os_dep/linux/os_intfs.o] Error 1

ecofintech, 2020-05-14:
uname -r
'5.4.0-29-generic'
Lubuntu 20.04

I have this error message at make

' ' '
  CC [M]  /home/user/rtl8188eus/os_dep/linux/os_intfs.o
/home/user/rtl8188eus/os_dep/linux/os_intfs.c:1240:22: error: initialization of ‘u16 (*)(struct net_device *, struct sk_buff *, struct net_device *)’ {aka ‘short unsigned int (*)(struct net_device *, struct sk_buff *, struct net_device *)’} from incompatible pointer type ‘u16 (*)(struct net_device *, struct sk_buff *, struct net_device *, u16 (*)(struct net_device *, struct sk_buff *, struct net_device *))’ {aka ‘short unsigned int (*)(struct net_device *, struct sk_buff *, struct net_device *, short unsigned int (*)(struct net_device *, struct sk_buff *, struct net_device *))’} [-Werror=incompatible-pointer-types]
 1240 |  .ndo_select_queue = rtw_select_queue,
      |                      ^~~~~~~~~~~~~~~~
/home/user/rtl8188eus/os_dep/linux/os_intfs.c:1240:22: note: (near initialization for ‘rtw_netdev_ops.ndo_select_queue’)
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:275: /home/user/rtl8188eus/os_dep/linux/os_intfs.o] Error 1
make[1]: *** [Makefile:1719: /home/user/rtl8188eus] Error 2
make[1]: se sale del directorio '/usr/src/linux-headers-5.4.0-29-generic'
make: *** [Makefile:1911: modules] Error 2
' ' '

Coolkkmeat, 2020-05-14:
I've run into this problem before, I'm a Linux novice.
---
# 1
Remember your version of the kernel first:
```
[root@localhost ~]# uname -a
Linux localhost 4.18.0-147.8.1.el8_1.x86_64 #1 SMP Thu Apr 9 13:49:54 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
```
For example, my kernel version is **4.18.0**

# 2
Then you need to modify two files:

os_dep/linux/os_intfs.c#L1102
```
#if (LINUX_VERSION_CODE>=KERNEL_VERSION(4,19,0))
Change the number here to your kernel version number, as follows.
#if (LINUX_VERSION_CODE>=KERNEL_VERSION(4,18,0))
```
os_dep/linux/rtw_android.c#L629
```
#if (LINUX_VERSION_CODE >= KERNEL_VERSION(5, 0, 0))
Same as the one before.
#if (LINUX_VERSION_CODE >= KERNEL_VERSION(4, 18, 0))
```
Then it can be compiled successfully.
I take my cue from this url: https://blog.csdn.net/mdykj33/article/details/103598386

Espinosa123321, 2020-06-06:
@Coolkkmeat i don't get it

Espinosa123321, 2020-06-06:
@Coolkkmeat arter i did that i tried to compile it again and i got this errors
`make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.6.0-kali2-amd64/build M=/root/rtl8188eus  modules
make[1]: Entering directory '/usr/src/linux-headers-5.6.0-kali2-amd64'
  CC [M]  /root/rtl8188eus/os_dep/linux/os_intfs.o
  CC [M]  /root/rtl8188eus/os_dep/linux/ioctl_cfg80211.o
/root/rtl8188eus/os_dep/linux/ioctl_cfg80211.c: In function ‘rtw_get_systime_us’:
/root/rtl8188eus/os_dep/linux/ioctl_cfg80211.c:353:18: error: storage size of ‘ts’ isn’t known
  353 |  struct timespec ts;
      |                  ^~
/root/rtl8188eus/os_dep/linux/ioctl_cfg80211.c:354:2: error: implicit declaration of function ‘getboottime’; did you mean ‘getboottime64’? [-Werror=implicit-function-declaration]
  354 |  getboottime(&ts);
      |  ^~~~~~~~~~~
      |  getboottime64
/root/rtl8188eus/os_dep/linux/ioctl_cfg80211.c:361:1: warning: control reaches end of non-void function [-Wreturn-type]
  361 | }
      | ^
/root/rtl8188eus/os_dep/linux/ioctl_cfg80211.c: In function ‘cfg80211_rtw_change_iface’:
/root/rtl8188eus/os_dep/linux/ioctl_cfg80211.c:1837:10: warning: this statement may fall through [-Wimplicit-fallthrough=]
 1837 |   is_p2p = _TRUE;
/root/rtl8188eus/os_dep/linux/ioctl_cfg80211.c:1839:2: note: here
 1839 |  case NL80211_IFTYPE_STATION:
      |  ^~~~
/root/rtl8188eus/os_dep/linux/ioctl_cfg80211.c:1861:10: warning: this statement may fall through [-Wimplicit-fallthrough=]
 1861 |   is_p2p = _TRUE;
/root/rtl8188eus/os_dep/linux/ioctl_cfg80211.c:1863:2: note: here
 1863 |  case NL80211_IFTYPE_AP:
      |  ^~~~
cc1: some warnings being treated as errors
make[3]: *** [/usr/src/linux-headers-5.6.0-kali2-common/scripts/Makefile.build:273: /root/rtl8188eus/os_dep/linux/ioctl_cfg80211.o] Error 1
make[2]: *** [/usr/src/linux-headers-5.6.0-kali2-common/Makefile:1704: /root/rtl8188eus] Error 2
make[1]: *** [/usr/src/linux-headers-5.6.0-kali2-common/Makefile:180: sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.6.0-kali2-amd64'
make: *** [Makefile:1911: modules] Error 2`

Coolkkmeat, 2020-06-06:
@Espinosa123321 I've never studied C, so I don't know. I'll try to search for related issues and here's my guess :
"storage size of 'ts' isn't known" seems to be missing some kind of header file.
"implicit declaration of function 'getboottime', maybe it should be changed to "getboottime64".

BlackCorp1, 2022-06-19:
alguem pode me ajudar por favor a resolver esses erros IRRITANTES do comando "make"??? sou novato no linux!!! quais comandos devo usar??? obrigado...

(root Black)-[-/rt18188eus] 
# make && make install make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.17.0-kali3-amd64/build M=/root/r tl8188eus modules make[1]: Entrando no diretório '/usr/src/linux-headers-5.17.0-kali3-amd64',
CC [M] /root/rtl8188eus/os_dep/osdep_service.o /root/rtl8188eus/os_dep/osdep service.c: In function 'thread_exit': /root/rtl8188eus/os_dep/osdep_service.c:1271:9: error: implicit declaration of fun ction 'complete_and_exit' [-Werror=implicit-function-declaration] 1271 |
complete_and_exit(comp, 0);
/root/rtl8188eus/os_dep/osdep_service.c: In function 'rtw_change_ifname': /root/rtl8188eus/os_dep/osdep_service.c:2450:28: warning: passing argument 1 of '. rtw_memcpy' discards 'const' qualifier from pointer target type [-Wdiscarded-quali, fiers] 2450 |
_rtw_memcpy(pnetdev-dev_addr, adapter_mac_addr(padapter), ETH_ALE
N);
/root/rtl8188eus/os_dep/osdep_service.c:899:24: note: expected 'void *' but argume int is of type 'const unsigned char *
899 | void _rtw_memcpy(void *dst, const void *src, u32 sz),
cc1: some warnings being treated as errors make[2]: *** [/usr/src/linux-headers-5.17.0-kali3-common/scripts/Makefile.build:29 3: /root/rtl8188eus/os_dep/osdep_service.o] Erro 1 make[1]: *** [/usr/src/linux-headers-5.17.0-kali3-common/Makefile:1855: /root/rtl8188eus] Erro 2 make[1]: Saindo do diretório '/usr/src/linux-headers-5.17.0-kali3-amd64' make: *** [Makefile:2058: modules] Erro 2