quickreflex/rtl8188eus #51: I had a lot of errors

Espinosa123321, 2020-06-05:
HELP!!!!! pls.
i already update everything in my linux, and i got this errors when i type make all 

make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.6.0-kali2-amd64/build M=/root/rtl8188eus  modules
make[1]: Entering directory '/usr/src/linux-headers-5.6.0-kali2-amd64'
  CC [M]  /root/rtl8188eus/core/rtw_cmd.o
  CC [M]  /root/rtl8188eus/core/rtw_security.o
  CC [M]  /root/rtl8188eus/core/rtw_debug.o
  CC [M]  /root/rtl8188eus/core/rtw_io.o
  CC [M]  /root/rtl8188eus/core/rtw_ioctl_query.o
  CC [M]  /root/rtl8188eus/core/rtw_ioctl_set.o
  CC [M]  /root/rtl8188eus/core/rtw_ieee80211.o
  CC [M]  /root/rtl8188eus/core/rtw_mlme.o
  CC [M]  /root/rtl8188eus/core/rtw_mlme_ext.o
/root/rtl8188eus/core/rtw_mlme_ext.c: In function ‘mgt_dispatcher’:
/root/rtl8188eus/core/rtw_mlme_ext.c:1441:6: warning: this statement may fall through [-Wimplicit-fallthrough=]
 1441 |   if (check_fwstate(pmlmepriv, WIFI_AP_STATE) == _TRUE)
      |      ^
/root/rtl8188eus/core/rtw_mlme_ext.c:1446:2: note: here
 1446 |  case WIFI_ASSOCREQ:
      |  ^~~~
  CC [M]  /root/rtl8188eus/core/rtw_mi.o
  CC [M]  /root/rtl8188eus/core/rtw_wlan_util.o
  CC [M]  /root/rtl8188eus/core/rtw_vht.o
  CC [M]  /root/rtl8188eus/core/rtw_pwrctrl.o
  CC [M]  /root/rtl8188eus/core/rtw_rf.o
  CC [M]  /root/rtl8188eus/core/rtw_recv.o
  CC [M]  /root/rtl8188eus/core/rtw_sta_mgt.o
  CC [M]  /root/rtl8188eus/core/rtw_ap.o
  CC [M]  /root/rtl8188eus/core/rtw_xmit.o
  CC [M]  /root/rtl8188eus/core/rtw_p2p.o
  CC [M]  /root/rtl8188eus/core/rtw_tdls.o
  CC [M]  /root/rtl8188eus/core/rtw_br_ext.o
  CC [M]  /root/rtl8188eus/core/rtw_iol.o
  CC [M]  /root/rtl8188eus/core/rtw_sreset.o
  CC [M]  /root/rtl8188eus/core/rtw_btcoex_wifionly.o
  CC [M]  /root/rtl8188eus/core/rtw_btcoex.o
  CC [M]  /root/rtl8188eus/core/rtw_beamforming.o
  CC [M]  /root/rtl8188eus/core/rtw_odm.o
  CC [M]  /root/rtl8188eus/core/efuse/rtw_efuse.o
  CC [M]  /root/rtl8188eus/os_dep/osdep_service.o
  CC [M]  /root/rtl8188eus/os_dep/linux/os_intfs.o
/root/rtl8188eus/os_dep/linux/os_intfs.c:1240:22: error: initialization of ‘u16 (*)(struct net_device *, struct sk_buff *, struct net_device *)’ {aka ‘short unsigned int (*)(struct net_device *, struct sk_buff *, struct net_device *)’} from incompatible pointer type ‘u16 (*)(struct net_device *, struct sk_buff *, struct net_device *, u16 (*)(struct net_device *, struct sk_buff *, struct net_device *))’ {aka ‘short unsigned int (*)(struct net_device *, struct sk_buff *, struct net_device *, short unsigned int (*)(struct net_device *, struct sk_buff *, struct net_device *))’} [-Werror=incompatible-pointer-types]
 1240 |  .ndo_select_queue = rtw_select_queue,
      |                      ^~~~~~~~~~~~~~~~
/root/rtl8188eus/os_dep/linux/os_intfs.c:1240:22: note: (near initialization for ‘rtw_netdev_ops.ndo_select_queue’)
cc1: some warnings being treated as errors
make[3]: *** [/usr/src/linux-headers-5.6.0-kali2-common/scripts/Makefile.build:273: /root/rtl8188eus/os_dep/linux/os_intfs.o] Error 1
make[2]: *** [/usr/src/linux-headers-5.6.0-kali2-common/Makefile:1704: /root/rtl8188eus] Error 2
make[1]: *** [/usr/src/linux-headers-5.6.0-kali2-common/Makefile:180: sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.6.0-kali2-amd64'
make: *** [Makefile:1911: modules] Error 2

and when i use make install this happens

install -p -m 644 8188eu.ko  /lib/modules/5.6.0-kali2-amd64/kernel/drivers/net/wireless/
install: cannot stat '8188eu.ko': No such file or directory
make: *** [Makefile:1917: install] Error 1

can somebody help

bobby69-24, 2020-06-23:
r u installed this driver previously! it worked or not

Espinosa123321, 2020-07-01:
I found out that that's not the driver

On Jul 1, 2020 7:55 PM, wrote:

I haven't

Espinosa123321, 2020-07-01:
I haven't

danwdart, 2020-07-29:
Same error here:

```
/home/dwd/code/other/rtl8188eus/os_dep/linux/os_intfs.c:1240:22: error: initialization of ‘u16 (*)(struct net_device *, struct sk_buff *, struct net_device *)’ {aka ‘short unsigned int (*)(struct net_device *, struct sk_buff *, struct net_device *)’} from incompatible pointer type ‘u16 (*)(struct net_device *, struct sk_buff *, struct net_device *, u16 (*)(struct net_device *, struct sk_buff *, struct net_device *))’ {aka ‘short unsigned int (*)(struct net_device *, struct sk_buff *, struct net_device *, short unsigned int (*)(struct net_device *, struct sk_buff *, struct net_device *))’} [-Werror=incompatible-pointer-types]
 1240 |  .ndo_select_queue = rtw_select_queue,
      |                      ^~~~~~~~~~~~~~~~
/home/dwd/code/other/rtl8188eus/os_dep/linux/os_intfs.c:1240:22: note: (near initialization for ‘rtw_netdev_ops.ndo_select_queue’)
  CC [M]  /home/dwd/code/other/rtl8188eus/os_dep/linux/xmit_linux.o
  CC [M]  /home/dwd/code/other/rtl8188eus/os_dep/linux/mlme_linux.o
  CC [M]  /home/dwd/code/other/rtl8188eus/os_dep/linux/recv_linux.o
  CC [M]  /home/dwd/code/other/rtl8188eus/os_dep/linux/ioctl_cfg80211.o
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:275: /home/dwd/code/other/rtl8188eus/os_dep/linux/os_intfs.o] Error 1
make[2]: *** Waiting for unfinished jobs....
/home/dwd/code/other/rtl8188eus/os_dep/linux/ioctl_cfg80211.c: In function ‘cfg80211_rtw_change_iface’:
/home/dwd/code/other/rtl8188eus/os_dep/linux/ioctl_cfg80211.c:1837:10: warning: this statement may fall through [-Wimplicit-fallthrough=]
 1837 |   is_p2p = _TRUE;
/home/dwd/code/other/rtl8188eus/os_dep/linux/ioctl_cfg80211.c:1839:2: note: here
 1839 |  case NL80211_IFTYPE_STATION:
      |  ^~~~
/home/dwd/code/other/rtl8188eus/os_dep/linux/ioctl_cfg80211.c:1861:10: warning: this statement may fall through [-Wimplicit-fallthrough=]
 1861 |   is_p2p = _TRUE;
/home/dwd/code/other/rtl8188eus/os_dep/linux/ioctl_cfg80211.c:1863:2: note: here
 1863 |  case NL80211_IFTYPE_AP:
      |  ^~~~
make[1]: *** [Makefile:1731: /home/dwd/code/other/rtl8188eus] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.4.0-42-generic'
make: *** [Makefile:1911: modules] Error 2
```

SajjadDaneshmand, 2020-12-08:
Hi. please fix the bug .my wireless is not working