quickreflex/rtl8188eus #52: Build Fails for Kernel 5.4.55-1-lts (x86) Arch Linux

randomguywithagun, 2020-08-08:
Used DKMS Installation Method with Kernel  5.4.55-1-lts (I know its not supported)
`
# dkms add .
# dkms build 8188eu/1.0 --kernelsourcedir /usr/src/linux-lts
`
didnt work, as apperently my kernel was 5.4.54-1 for some reason, while my headers where 5.4.55-1 so i did.
`# pacman -Syu linux-lts linux-lts-headers`

And got this error in  /var/lib/dkms/8188eu/1.0/build/make.log

`DKMS make.log for 8188eu-1.0 for kernel 5.4.55-1-lts (x86_64)
Sa 8. Aug 13:48:35 CEST 2020
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.4.55-1-lts/build M=/var/lib/dkms/8188eu/1.0/build  modules
make[1]: Verzeichnis „/usr/lib/modules/5.4.55-1-lts/build“ wird betreten
  CC [M]  /var/lib/dkms/8188eu/1.0/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/8188eu/1.0/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/8188eu/1.0/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/8188eu/1.0/build/core/rtw_io.o
  CC [M]  /var/lib/dkms/8188eu/1.0/build/core/rtw_ioctl_query.o
  CC [M]  /var/lib/dkms/8188eu/1.0/build/core/rtw_ioctl_set.o
  CC [M]  /var/lib/dkms/8188eu/1.0/build/core/rtw_ieee80211.o
  CC [M]  /var/lib/dkms/8188eu/1.0/build/core/rtw_mlme.o
  CC [M]  /var/lib/dkms/8188eu/1.0/build/core/rtw_mlme_ext.o
/var/lib/dkms/8188eu/1.0/build/core/rtw_mlme_ext.c: In Funktion »mgt_dispatcher«:
/var/lib/dkms/8188eu/1.0/build/core/rtw_mlme_ext.c:1441:6: Warnung: diese Anweisung könnte durchfallen [-Wimplicit-fallthrough=]
 1441 |   if (check_fwstate(pmlmepriv, WIFI_AP_STATE) == _TRUE)
      |      ^
/var/lib/dkms/8188eu/1.0/build/core/rtw_mlme_ext.c:1446:2: Anmerkung: hier
 1446 |  case WIFI_ASSOCREQ:
      |  ^~~~
  CC [M]  /var/lib/dkms/8188eu/1.0/build/core/rtw_mi.o
  CC [M]  /var/lib/dkms/8188eu/1.0/build/core/rtw_wlan_util.o
  CC [M]  /var/lib/dkms/8188eu/1.0/build/core/rtw_vht.o
  CC [M]  /var/lib/dkms/8188eu/1.0/build/core/rtw_pwrctrl.o
  CC [M]  /var/lib/dkms/8188eu/1.0/build/core/rtw_rf.o
  CC [M]  /var/lib/dkms/8188eu/1.0/build/core/rtw_recv.o
  CC [M]  /var/lib/dkms/8188eu/1.0/build/core/rtw_sta_mgt.o
  CC [M]  /var/lib/dkms/8188eu/1.0/build/core/rtw_ap.o
  CC [M]  /var/lib/dkms/8188eu/1.0/build/core/rtw_xmit.o
  CC [M]  /var/lib/dkms/8188eu/1.0/build/core/rtw_p2p.o
  CC [M]  /var/lib/dkms/8188eu/1.0/build/core/rtw_tdls.o
  CC [M]  /var/lib/dkms/8188eu/1.0/build/core/rtw_br_ext.o
  CC [M]  /var/lib/dkms/8188eu/1.0/build/core/rtw_iol.o
  CC [M]  /var/lib/dkms/8188eu/1.0/build/core/rtw_sreset.o
  CC [M]  /var/lib/dkms/8188eu/1.0/build/core/rtw_btcoex_wifionly.o
  CC [M]  /var/lib/dkms/8188eu/1.0/build/core/rtw_btcoex.o
  CC [M]  /var/lib/dkms/8188eu/1.0/build/core/rtw_beamforming.o
  CC [M]  /var/lib/dkms/8188eu/1.0/build/core/rtw_odm.o
  CC [M]  /var/lib/dkms/8188eu/1.0/build/core/efuse/rtw_efuse.o
  CC [M]  /var/lib/dkms/8188eu/1.0/build/os_dep/osdep_service.o
  CC [M]  /var/lib/dkms/8188eu/1.0/build/os_dep/linux/os_intfs.o
/var/lib/dkms/8188eu/1.0/build/os_dep/linux/os_intfs.c:1240:22: Fehler: Initialisierung von »u16 (*)(struct net_device *, struct sk_buff *, struct net_device *)« {alias »short unsigned int (*)(struct net_device *, struct sk_buff *, struct net_device *)«} von inkompatiblem Zeigertyp »u16 (*)(struct net_device *, struct sk_buff *, struct net_device *, u16 (*)(struct net_device *, struct sk_buff *, struct net_device *))« {alias »short unsigned int (*)(struct net_device *, struct sk_buff *, struct net_device *, short unsigned int (*)(struct net_device *, struct sk_buff *, struct net_device *))«} [-Werror=incompatible-pointer-types]
 1240 |  .ndo_select_queue = rtw_select_queue,
      |                      ^~~~~~~~~~~~~~~~
/var/lib/dkms/8188eu/1.0/build/os_dep/linux/os_intfs.c:1240:22: Anmerkung: (nahe der Initialisierung für »rtw_netdev_ops.ndo_select_queue«)
cc1: Einige Warnungen werden als Fehler behandelt
make[2]: *** [scripts/Makefile.build:266: /var/lib/dkms/8188eu/1.0/build/os_dep/linux/os_intfs.o] Fehler 1
make[1]: *** [Makefile:1706: /var/lib/dkms/8188eu/1.0/build] Fehler 2
make[1]: Verzeichnis „/usr/lib/modules/5.4.55-1-lts/build“ wird verlassen
make: *** [Makefile:1911: modules] Fehler 2`

Thank you quickreflex for your work. :)

Ryazbeck, 2020-09-03:
I think my error is the same.

>Linux black-pearl 5.4.51+ #1333 Mon Aug 10 16:38:02 BST 2020 armv6l GNU/Linux

> $ tail /var/lib/dkms/8188eu/1.0/build/make.log
>   CC [M]  /var/lib/dkms/8188eu/1.0/build/os_dep/linux/os_intfs.o
> /var/lib/dkms/8188eu/1.0/build/os_dep/linux/os_intfs.c:1240:22: error: initialization of ‘u16 (*)(struct net_device *, struct sk_buff *, struct net_device *)’ {aka ‘short unsigned int (*)(struct net_device *, struct sk_buff *, struct net_device *)’} from incompatible pointer type ‘u16 (*)(struct net_device *, struct sk_buff *, struct net_device *, u16 (*)(struct net_device *, struct sk_buff *, struct net_device *))’ {aka ‘short unsigned int (*)(struct net_device *, struct sk_buff *, struct net_device *, short unsigned int (*)(struct net_device *, struct sk_buff *, struct net_device *))’} [-Werror=incompatible-pointer-types]
>   .ndo_select_queue = rtw_select_queue,
>                       ^~~~~~~~~~~~~~~~
> /var/lib/dkms/8188eu/1.0/build/os_dep/linux/os_intfs.c:1240:22: note: (near initialization for ‘rtw_netdev_ops.ndo_select_queue’)
> cc1: some warnings being treated as errors
> make[2]: *** [scripts/Makefile.build:266: /var/lib/dkms/8188eu/1.0/build/os_dep/linux/os_intfs.o] Error 1
> make[1]: *** [Makefile:1709: /var/lib/dkms/8188eu/1.0/build] Error 2
> make[1]: Leaving directory '/usr/src/linux-headers-5.4.51+'
> make: *** [Makefile:1911: modules] Error 2

Ryazbeck, 2020-09-03:
same error  https://github.com/quickreflex/rtl8188eus/issues/51

datalot-369, 2020-09-04:
Replicated in Linux Mint 20, impossible to do a `make all`.
I needed to change to Kernel 5.1. Same thing there.

What I tried was compiling in my MX Linux 19.2 system because [this launchpad report](because of https://bugs.launchpad.net/ubuntu/+source/gcc-9/+bug/1830961).
MX Linux is pretty much "stable/older" so it would compile well due to the GCC version difference and effectively it did compile over Linux 4.19.

What I want to try now is to install that 5.1 Kernel in MX Linux then compile and then do the `make install` step in Linux Mint.

Ryazbeck, 2020-09-05:
I was able to get my rtl8188eus dongle working with this driver https://github.com/aircrack-ng/rtl8188eus