quickreflex/rtl8188eus #53: Raspberry Pi, Kali and TP-LINK TL-WN722N (EU) Version 3.20

TheBallOfAeolus, 2020-09-29:
Hi everybody
I have a Raspberry Pi 2 Model B Rev 1.1 and a TP-LINK TL-WN722N (EU) Ver.:3.20 (150 Mbps High Gain Wireless USB Adapter - White)

Installed:
OS: Kali Linux 
Kernel: armv7l Linux 4.19.118-Re4son-v7+
CPU: ARMv7 rev 5 (v7l) @ 4x 900MHz

USB:
```
:~$ sudo lshw -C network
  *-network:0
       description: Wireless interface
       physical id: 2
       bus info: usb@1:1.5
       logical name: wlan0
       serial: d0:37:45:0e:6b:1a
       capabilities: ethernet physical wireless
       configuration: broadcast=yes driver=r8188eu multicast=yes wireless=unassociated
```

Compiling:
```
:~$ git clone https://github.com/quickreflex/rtl8188eus.git
Cloning into 'rtl8188eus'...
remote: Enumerating objects: 1312, done.
remote: Total 1312 (delta 0), reused 0 (delta 0), pack-reused 1312
Receiving objects: 100% (1312/1312), 4.64 MiB | 2.78 MiB/s, done.
Resolving deltas: 100% (777/777), done.
Updating files: 100% (588/588), done.

:~$ cd rtl8188eus/

:~$ make all
make ARCH=armv7l CROSS_COMPILE= -C /lib/modules/4.19.118-Re4son-v7+/build M=/home/hwit/dev/rtl8188eus  modules
make[1]: Entering directory '/usr/src/linux-headers-4.19.118-Re4son-v7+'
Makefile:614: arch/armv7l/Makefile: No such file or directory
make[1]: *** No rule to make target 'arch/armv7l/Makefile'.  Stop.
make[1]: Leaving directory '/usr/src/linux-headers-4.19.118-Re4son-v7+'
make: *** [Makefile:1911: modules] Error 2
```

becuase of the `Makefile:614: arch/armv7l/Makefile: No such file or directory` I tried instead:
```
:~$ make all ARCH=arm
make ARCH=arm CROSS_COMPILE= -C /lib/modules/4.19.118-Re4son-v7+/build M=/home/hwit/dev/rtl8188eus  modules
make[1]: Entering directory '/usr/src/linux-headers-4.19.118-Re4son-v7+'
  CC [M]  /home/hwit/dev/rtl8188eus/core/rtw_cmd.o
  CC [M]  /home/hwit/dev/rtl8188eus/core/rtw_security.o
  CC [M]  /home/hwit/dev/rtl8188eus/core/rtw_debug.o
  CC [M]  /home/hwit/dev/rtl8188eus/core/rtw_io.o
  CC [M]  /home/hwit/dev/rtl8188eus/core/rtw_ioctl_query.o
  CC [M]  /home/hwit/dev/rtl8188eus/core/rtw_ioctl_set.o
  CC [M]  /home/hwit/dev/rtl8188eus/core/rtw_ieee80211.o
  CC [M]  /home/hwit/dev/rtl8188eus/core/rtw_mlme.o
  CC [M]  /home/hwit/dev/rtl8188eus/core/rtw_mlme_ext.o
  CC [M]  /home/hwit/dev/rtl8188eus/core/rtw_mi.o
  CC [M]  /home/hwit/dev/rtl8188eus/core/rtw_wlan_util.o
  CC [M]  /home/hwit/dev/rtl8188eus/core/rtw_vht.o
  CC [M]  /home/hwit/dev/rtl8188eus/core/rtw_pwrctrl.o
  CC [M]  /home/hwit/dev/rtl8188eus/core/rtw_rf.o
  CC [M]  /home/hwit/dev/rtl8188eus/core/rtw_recv.o
  CC [M]  /home/hwit/dev/rtl8188eus/core/rtw_sta_mgt.o
  CC [M]  /home/hwit/dev/rtl8188eus/core/rtw_ap.o
/home/hwit/dev/rtl8188eus/core/rtw_ap.c: In function ‘rtw_add_bcn_ie’:
/home/hwit/dev/rtl8188eus/core/rtw_ap.c:193:9: warning: ‘ielen’ may be used uninitialized in this function [-Wmaybe-uninitialized]
  193 |   ielen += 2;
      |   ~~~~~~^~~~
  CC [M]  /home/hwit/dev/rtl8188eus/core/rtw_xmit.o
  CC [M]  /home/hwit/dev/rtl8188eus/core/rtw_p2p.o
[... 141 lines cut ...]
uio_pdrv_genirq        16384  0
uio                    20480  1 uio_pdrv_genirq
fixed                  16384  0
drm                   442368  0
drm_panel_orientation_quirks    16384  1 drm
ip_tables              24576  0
x_tables               32768  1 ip_tables
ipv6                  458752  34
```

...of course but I had to try:
```
:~$ sudo ifconfig wlan0 down
:~$ sudo airmon-ng check kill
Killing these processes:

  PID Name
  368 dhclient
  509 wpa_supplicant

:~$ sudo iw dev wlan0 set type managed
command failed: No such device (-19)
```

Any suggestion?

TheBallOfAeolus, 2020-10-04:
Additional info
```
:~$ sudo lsusb | grep TL-WN722N
Bus 001 Device 005: ID 2357:010c TP-Link TL-WN722N v2/v3 [Realtek RTL8188EUS]
:~$  grep v2357p010C /lib/modules/`uname -r`/modules.alias
alias usb:v2357p010Cd*dc*dsc*dp*ic*isc*ip*in* 8188eu
alias usb:v2357p010Cd*dc*dsc*dp*ic*isc*ip*in* r8188eu
:~$  cat /etc/modprobe.d/realtek.conf
blacklist r8188eu
```

SisyphusIsntHappy, 2021-08-12:
Builtin kernel driver for this chip-set is not equipped for monitor mode, you need separate driver by the aircrack-ng.

GetRektBoy724, 2021-08-13:
> Hi everybody
> I have a Raspberry Pi 2 Model B Rev 1.1 and a TP-LINK TL-WN722N (EU) Ver.:3.20 (150 Mbps High Gain Wireless USB Adapter - White)
> 
> Installed:
> OS: Kali Linux
> Kernel: armv7l Linux 4.19.118-Re4son-v7+
> CPU: ARMv7 rev 5 (v7l) @ 4x 900MHz
> 
> USB:
> 
> ```
> :~$ sudo lshw -C network
>   *-network:0
>        description: Wireless interface
>        physical id: 2
>        bus info: usb@1:1.5
>        logical name: wlan0
>        serial: d0:37:45:0e:6b:1a
>        capabilities: ethernet physical wireless
>        configuration: broadcast=yes driver=r8188eu multicast=yes wireless=unassociated
> ```
> 
> Compiling:
> 
> ```
> :~$ git clone https://github.com/quickreflex/rtl8188eus.git
> Cloning into 'rtl8188eus'...
> remote: Enumerating objects: 1312, done.
> remote: Total 1312 (delta 0), reused 0 (delta 0), pack-reused 1312
> Receiving objects: 100% (1312/1312), 4.64 MiB | 2.78 MiB/s, done.
> Resolving deltas: 100% (777/777), done.
> Updating files: 100% (588/588), done.
> 
> :~$ cd rtl8188eus/
> 
> :~$ make all
> make ARCH=armv7l CROSS_COMPILE= -C /lib/modules/4.19.118-Re4son-v7+/build M=/home/username/dev/rtl8188eus  modules
> make[1]: Entering directory '/usr/src/linux-headers-4.19.118-Re4son-v7+'
> Makefile:614: arch/armv7l/Makefile: No such file or directory
> make[1]: *** No rule to make target 'arch/armv7l/Makefile'.  Stop.
> make[1]: Leaving directory '/usr/src/linux-headers-4.19.118-Re4son-v7+'
> make: *** [Makefile:1911: modules] Error 2
> ```
> 
> becuase of the `Makefile:614: arch/armv7l/Makefile: No such file or directory` I tried instead:
> 
> ```
> :~$ make all ARCH=arm
> make ARCH=arm CROSS_COMPILE= -C /lib/modules/4.19.118-Re4son-v7+/build M=/home/username/dev/rtl8188eus  modules
> make[1]: Entering directory '/usr/src/linux-headers-4.19.118-Re4son-v7+'
>   CC [M]  /home/username/dev/rtl8188eus/core/rtw_cmd.o
>   CC [M]  /home/username/dev/rtl8188eus/core/rtw_security.o
>   CC [M]  /home/username/dev/rtl8188eus/core/rtw_debug.o
>   CC [M]  /home/username/dev/rtl8188eus/core/rtw_io.o
>   CC [M]  /home/username/dev/rtl8188eus/core/rtw_ioctl_query.o
>   CC [M]  /home/username/dev/rtl8188eus/core/rtw_ioctl_set.o
>   CC [M]  /home/username/dev/rtl8188eus/core/rtw_ieee80211.o
>   CC [M]  /home/username/dev/rtl8188eus/core/rtw_mlme.o
>   CC [M]  /home/username/dev/rtl8188eus/core/rtw_mlme_ext.o
>   CC [M]  /home/username/dev/rtl8188eus/core/rtw_mi.o
>   CC [M]  /home/username/dev/rtl8188eus/core/rtw_wlan_util.o
>   CC [M]  /home/username/dev/rtl8188eus/core/rtw_vht.o
>   CC [M]  /home/username/dev/rtl8188eus/core/rtw_pwrctrl.o
>   CC [M]  /home/username/dev/rtl8188eus/core/rtw_rf.o
>   CC [M]  /home/username/dev/rtl8188eus/core/rtw_recv.o
>   CC [M]  /home/username/dev/rtl8188eus/core/rtw_sta_mgt.o
>   CC [M]  /home/username/dev/rtl8188eus/core/rtw_ap.o
> /home/username/dev/rtl8188eus/core/rtw_ap.c: In function ‘rtw_add_bcn_ie’:
> /home/username/dev/rtl8188eus/core/rtw_ap.c:193:9: warning: ‘ielen’ may be used uninitialized in this function [-Wmaybe-uninitialized]
>   193 |   ielen += 2;
[... 153 lines cut ...]
> ipv6                  458752  34
> ```
> 
> ...but monitor mode is not working! :(
> 
> ```
> :~$ sudo ifconfig wlan0 down
> :~$ sudo airmon-ng check kill
> Killing these processes:
> 
>   PID Name
>   368 dhclient
>   509 wpa_supplicant
> 
> :~$ sudo iw dev wlan0 set type managed
> command failed: No such device (-19)
> ```
> 
> Any suggestion?

no, you don't need to manually compile this driver, if you're in Kali Linux, you can do
```
apt-get update
apt-get install realtek-rtl8188eus-dkms
```

SisyphusIsntHappy, 2021-08-13:
Hi, 

I can't increase tx power for this chipset using. I'm getting a 

`Error for wireless request "Set Tx Power" (8B26) :
    SET failed on device wlp0s20u2 ; Operation not permitted.`
    
    Would you know anything about that?