quickreflex/rtl8188eus #55: nearly installed i presume

GarethHards, 2021-02-03:
make install && make modules_install
install -p -m 644 8188eu.ko  /lib/modules/5.10.11-v7l+/kernel/drivers/net/wireless/
install: cannot stat '8188eu.ko': No such file or directory
make: *** [Makefile:1917: install] Error 1

GarethHards, 2021-02-03:
pi@raspberrypi:~/rtl8188eus $ ls
clean      ifcfg-wlan0  Makefile   rtl8188EUS_linux_v4.1.4_6773.20130222
core       include      os_dep     runwpa
dkms.conf  Kconfig      platform   wlan0dhcp
hal        LICENSE      README.md  wpa_0_8.conf
pi@raspberrypi:~/rtl8188eus $ make all
make ARCH=armv7l CROSS_COMPILE= -C /lib/modules/5.10.11-v7l+/build M=/home/pi/rtl8188eus  modules
make[1]: *** /lib/modules/5.10.11-v7l+/build: No such file or directory.  Stop.
make: *** [Makefile:1911: modules] Error 2
pi@raspberrypi:~/rtl8188eus $  ./build-driver
bash: ./build-driver: No such file or directory
pi@raspberrypi:~/rtl8188eus $ make install && make modules_install
install -p -m 644 8188eu.ko  /lib/modules/5.10.11-v7l+/kernel/drivers/net/wireless/
install: cannot stat '8188eu.ko': No such file or directory
make: *** [Makefile:1917: install] Error 1
pi@raspberrypi:~/rtl8188eus $ make && make modules && make install && make modules_install
make ARCH=armv7l CROSS_COMPILE= -C /lib/modules/5.10.11-v7l+/build M=/home/pi/rtl8188eus  modules
make[1]: *** /lib/modules/5.10.11-v7l+/build: No such file or directory.  Stop.
make: *** [Makefile:1911: modules] Error 2
pi@raspberrypi:~/rtl8188eus $ zcat /proc/config.gz > .config &&  make && make modules && make install && make modules_install
gzip: /proc/config.gz: No such file or directory
pi@raspberrypi:~/rtl8188eus $ make ARCH={TARGET-ARCHITERCTURE} CROSS_COMPILE={COMPILER}; make ARCH={TARGET-ARCHITERCTURE} CROSS_COMPILE={COMPILER} modules && make install && make modules_install
make ARCH={TARGET-ARCHITERCTURE} CROSS_COMPILE={COMPILER} -C /lib/modules/5.10.11-v7l+/build M=/home/pi/rtl8188eus  modules
make[1]: *** /lib/modules/5.10.11-v7l+/build: No such file or directory.  Stop.
make: *** [Makefile:1911: modules] Error 2
make ARCH={TARGET-ARCHITERCTURE} CROSS_COMPILE={COMPILER} -C /lib/modules/5.10.11-v7l+/build M=/home/pi/rtl8188eus  modules
make[1]: *** /lib/modules/5.10.11-v7l+/build: No such file or directory.  Stop.
make: *** [Makefile:1911: modules] Error 2
pi@raspberrypi:~/rtl8188eus $ make modules_install install
make: *** No rule to make target 'modules_install'.  Stop.
pi@raspberrypi:~/rtl8188eus $ make-kpkg --initrd kernel_image
bash: make-kpkg: command not found
pi@raspberrypi:~/rtl8188eus $ LS
bash: LS: command not found
pi@raspberrypi:~/rtl8188eus $ ls
clean  core  dkms.conf  hal  ifcfg-wlan0  include  Kconfig  LICENSE  Makefile  os_dep  platform  README.md  rtl8188EUS_linux_v4.1.4_6773.20130222  runwpa  wlan0dhcp  wpa_0_8.conf
pi@raspberrypi:~/rtl8188eus $ sudo nano kconfig
pi@raspberrypi:~/rtl8188eus $

GarethHards, 2021-02-03:
what do i do next?

aninternetian, 2021-04-29:
switch to root