quickreflex/rtl8188eus #56: Bad return status for module build on kernel

TheGreatHalva, 2021-03-29:
Ran these commands: 
```bash 
➜  rtl8188eus git:(v5.2.2.4) sudo dkms add .

Creating symlink /var/lib/dkms/8188eu/1.0/source ->
                 /usr/src/8188eu-1.0

DKMS: add completed.
```
```bash
➜  rtl8188eus git:(v5.2.2.4) sudo dkms build 8188eu/1.0 

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make' all KVER=5.8.0-7642-generic............(bad exit status: 2)
ERROR (dkms apport): binary package for 8188eu: 1.0 not found
Error! Bad return status for module build on kernel: 5.8.0-7642-generic (x86_64)
Consult /var/lib/dkms/8188eu/1.0/build/make.log for more information.
```
The make.log tells this:
```
In file included from /var/lib/dkms/8188eu/1.0/build/include/drv_types.h:82,
                 from /var/lib/dkms/8188eu/1.0/build/core/rtw_br_ext.c:32:
/var/lib/dkms/8188eu/1.0/build/include/rtw_security.h:249:8: error: redefinition of ‘struct sha256_state’
  249 | struct sha256_state {
      |        ^~~~~~~~~~~~
In file included from ./include/linux/filter.h:23,
                 from ./include/net/sock.h:59,
                 from ./include/net/inet_sock.h:22,
                 from ./include/net/ip.h:27,
                 from /var/lib/dkms/8188eu/1.0/build/core/rtw_br_ext.c:24:
./include/crypto/sha.h:84:8: note: originally defined here
   84 | struct sha256_state {
      |        ^~~~~~~~~~~~
make[2]: *** [scripts/Makefile.build:286: /var/lib/dkms/8188eu/1.0/build/core/rtw_br_ext.o] Error 1
make[1]: *** [Makefile:1783: /var/lib/dkms/8188eu/1.0/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.8.0-7642-generic'
make: *** [Makefile:1911: modules] Error 2
```
OS: Pop!_OS 20.10 (based on Ubuntu 20.10)

lucas-FP, 2021-03-31:
Same issue here

TheGreatHalva, 2021-04-02:
nope. using built-in driver right now, but the performance on my TP-Link adapter is far from good. 
funnilly enough, this driver worked perfectly on arch. may this be one of ubuntu-specific bugs?

listenerri, 2021-06-23:
+1 Manjaro with kernel 5.10.42

listenerri, 2021-06-23:
Similar issue:
https://github.com/aircrack-ng/rtl8188eus/issues/73