quickreflex/rtl8188eus #60: Failed to build

marek22k, 2022-10-21:
```
$make
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/6.0.0-2parrot1-amd64/build M=/home/marek/Downloads/rtl8188eus  modules
make[1]: Entering directory '/usr/src/linux-headers-6.0.0-2parrot1-amd64'
  CC [M]  /home/marek/Downloads/rtl8188eus/core/rtw_cmd.o
In file included from /home/marek/Downloads/rtl8188eus/include/osdep_service.h:46,
                 from /home/marek/Downloads/rtl8188eus/include/drv_types.h:32,
                 from /home/marek/Downloads/rtl8188eus/core/rtw_cmd.c:22:
/home/marek/Downloads/rtl8188eus/core/rtw_cmd.c: In function ‘rtw_cmd_thread’:
/home/marek/Downloads/rtl8188eus/include/osdep_service_linux.h:185:23: error: implicit declaration of function ‘complete_and_exit’ [-Werror=implicit-function-declaration]
  185 | #define thread_exit() complete_and_exit(NULL, 0)
      |                       ^~~~~~~~~~~~~~~~~
/home/marek/Downloads/rtl8188eus/core/rtw_cmd.c:747:9: note: in expansion of macro ‘thread_exit’
  747 |         thread_exit();
      |         ^~~~~~~~~~~
/home/marek/Downloads/rtl8188eus/core/rtw_cmd.c:749:1: error: control reaches end of non-void function [-Werror=return-type]
  749 | }
      | ^
cc1: some warnings being treated as errors
make[2]: *** [/usr/src/linux-headers-6.0.0-2parrot1-common/scripts/Makefile.build:254: /home/marek/Downloads/rtl8188eus/core/rtw_cmd.o] Error 1
make[1]: *** [/usr/src/linux-headers-6.0.0-2parrot1-common/Makefile:1876: /home/marek/Downloads/rtl8188eus] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.0.0-2parrot1-amd64'
make: *** [Makefile:1911: modules] Error 2
```
OS: Parrot OS 
```
$uname -a
Linux parrot 6.0.0-2parrot1-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.0.2-1parrot1 (2022-10-18) x86_64 GNU/Linux
```

mat-vc, 2022-10-21:
same issue here, kernel 6.0.0-2parrot1-amd64

marek22k, 2022-10-21:
https://github.com/lwfinger/rtl8188eu seems to work